The-OpenROAD-Project/OpenLane #1: Change submodule command to do shallow command

mithro, 2020-07-20:
It is probably a good idea to clone the library submodules in "shallow" mode.

```
    git submodule update --init libraries/sky130_fd_sc_hd/latest
```

See related issue @ https://github.com/google/skywater-pdk/issues/108

ax3ghazy, 2020-07-31:
@mithro: Since they're properly configured on the skywater-pdk repo as shallow submodules (e.g., https://github.com/google/skywater-pdk/blob/master/.gitmodules#L6), the basic submodule command you mentioned does already clone the submodules shallowly, so it already passes `--depth 1` to the clone command.

Cloning the skywater-pdk itself shallowly doesn't provide much gain; it only saves like 400 KB and then this https://github.com/google/skywater-pdk/blob/master/scripts/make/git.mk wouldn't work because it is looking for latest tag, which is lost in the case of a shallow clone.

riking, 2020-08-08:
Now need to update scripts/make/git.mk now that that's the instructions.

closed 2021-06-16 by donn