The-OpenROAD-Project/OpenLane #5: no such file or directory Issue

mdzakirhussain, 2020-07-25:
I had followed every single step given, but I am facing the following Issue.

[INFO]: Version: rc1
[INFO]: Using design configuration at /openLANE_flow/designs/spm/config.tcl
couldn't read file "/openLANE_flow/pdks//sky130A/libs.tech/openlane/config.tcl": no such file or directory
    while executing
"source $pdk_config"
    (procedure "prep" line 86)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 7)
    invoked from within
"run_non_interactive_mode {*}$argv_copy"
    invoked from within
"if { [info exists flags_map(-interactive)] ||\
    [info exists flags_map(-it)] } {
	if { [info exists arg_values(-file)] } {
		run_file [file normali..."
    (file "./flow.tcl" line 136)

20Mhz, 2020-07-25:
Well, there is suppose to be a file there, is /openLANE_flow your actual work directory or is it missing a prefix? did you successfully build open_pdk?

mdzakirhussain, 2020-07-25:
During open-pdk

While performing the following command 

make

I got the following some messages

Error message output from magic:
Can't write file named '(UNNAMED)'

sky130_fd_sc_hd__buf_12_1"
LEF read:  Further warnings will not be reported.
Warning: Cannot open file for writing!
Warning: Cannot open file for writing!
Warning: Cannot open file for writing!
Warning: Cannot open file for writing!

PS: I already have magic installed in my system.

20Mhz, 2020-07-26:
I think the UNNAMED is unrelated, but Warning may be related to wrong path somewhere

mdzakirhussain, 2020-07-26:
Ok. But, how do I solve the issue, I just followed the steps given.

smunaut, 2020-07-26:
So the `Warning: Cannot open file for writing!` (and a bunch of other warnings during open_pdk) build can be traced to open_pdk reading both the `.lef` and the `.magic.lef` files during aggregation resulting in duplicates.

agorararmard, 2020-07-27:
@mdzakirhussain are you sure this is your directory structure for rc1?

openlane_working_dir
├── openlane
├── pdks
├──── skywater-pdk
├──── open_pdks
├────── sky130
├──────── pdks
├───────── sky130A

Meaning that both skywater-pdk and open_pdks are under pdks. This should've been clearer in the README and hopefully this is the case in rc2 that was just released yesterday.

closed 2020-07-29 by agorararmard