The-OpenROAD-Project/OpenLane #9: Cannot install:  Docker error

RTimothyEdwards, 2020-07-26:
I tried to install openlane (branch rc1) following the instructions (cd openlane/docker_build ; make merge).  But the build always fails with:

    Step 2/22 : RUN yum install -y git tcl tk libjpeg libgomp libXext libSM libXft libffi cairo gettext Xvfb
       ---> Running in 4f9707bae11a
   The command '/bin/sh -c yum install -y git tcl tk libjpeg libgomp libXext libSM libXft libffi cairo gettext Xvfb' returned a non-zero code: 139

Any clue as to what is going on here?

RTimothyEdwards, 2020-07-26:
Solution provided by @yrappt on the skywater-pdk slack workspace:  This issue is from the spectre attack vulnerability in Intel processors, oddly enough, and there is a solution mentioned at the bottom of the page here:  https://github.com/CentOS/sig-cloud-instance-images/issues/103 , where the solution is to enable vsyscall at boot time by editing the grub config and rebooting.  Worked like a charm.

bzboi, 2020-07-28:
Hi, I'm running Ubuntu 20.04 on WSL2, is the edit made to grub in Ubuntu or in Centos? I tried "sudo nano /etc/default/grub" followed by "sudo update-grub" in Ubuntu and rebooted but am still seeing the same error. I'm a Docker noob and will need help if the edit needs to be made within Centos. Thanks.

[image]

RTimothyEdwards, 2020-07-28:
@bzboi :  There was a second solution that I had written in my notes that I had used earlier on an openlane install, which was to change "centos6" to "centos7" in the dockerfile.  There are some issues with using centos7, but the only one I found was that yosys fails in centos7 due to a broken shared library reference.  I made a symbolic link "/lib64/libffi.so.5" pointing to libffi.so.6 to fix that.  After that, I was able to run the flow on a design.  I have no idea if other centos7-compatibility problems might happen if you do that.

As for the above solution, I'm doubtful that "grub" actually does anything under WSL, since Windows does the booting, not Ubuntu.

RTimothyEdwards, 2020-07-28:
I am reopening this issue, as @bzboi has an active and unsolved variant of this issue.

nurirfansyah, 2020-11-17:
> Hi, I'm running Ubuntu 20.04 on WSL2, is the edit made to grub in Ubuntu or in Centos? I tried "sudo nano /etc/default/grub" followed by "sudo update-grub" in Ubuntu and rebooted but am still seeing [...]

Hi, I had the same problem, I am running Ubuntu 18.04 on WSL2. I tried the steps to enable vsyscall=emulate as mentioned here: https://github.com/microsoft/WSL/issues/4694, which is to have a .wslconfig file with the following lines:

[wsl2]
kernelCommandLine = vsyscall=emulate

This worked for me, I am now able to build the Openlane docker in Ubuntu on WSL2.

agorararmard, 2020-12-07:
@RTimothyEdwards, @bzboi, and @nurirfansyah: We have migrated from using centos6 to centos7 in openlane master. We also provided an alternative for building the docker locally, pulling a pre-built image from dockerhub. Does this help close this long-standing issue?

donn, 2021-06-16:
Given that people haven't complained in a while, I'm going to assume this issue is resolved.

closed 2021-06-16 by donn