The-OpenROAD-Project/OpenLane #10: Netgen hangs

20Mhz, 2020-07-27:
Tool hangs trying to resolve symetries 

```
  Class: sky130_fd_sc_hd__fill_2 instances: 26035
  Class: sky130_fd_sc_hd__decap_12 instances: 6555
  Class: sky130_fd_sc_hd__nor2_4 instances: 1057
Circuit contains 12008 nets, and 5 disconnected pins.

Circuit 1 contains 106880 devices, Circuit 2 contains 106880 devices.
Circuit 1 contains 12008 nets,    Circuit 2 contains 12008 nets.

Circuits match with 5586 symmetries.
Netlists match with 5586 symmetries.
```

agorararmard, 2020-07-27:
@20Mhz LVS usually takes a long time on big designs. You might want to comment out  run_netgen from [flow.tcl](https://github.com/efabless/openlane/blob/master/flow.tcl) until you perfect the design (get down to 0 DRC violations), then put it back.

agorararmard, 2020-09-28:
Merged to master.

closed 2020-09-28 by agorararmard