The-OpenROAD-Project/OpenLane #16: Explore other base images for the docker option (e.g., newer centos versions, ubuntu, debian)

ax3ghazy, 2020-07-28:
Not a good idea to keep telling people to modify their boot loader configuration. See https://github.com/efabless/openlane/issues/9 for details.

Should help with running natively as well.

nadeemyaseen-lm, 2021-01-07:
I am working on building openLANE image based on [Ubuntu OS](https://hub.docker.com/_/swift) and will try to integrate [FAULT](https://github.com/Cloud-V/Fault) tool as well.

agorararmard, 2021-01-08:
@nadeemyaseen-lm: In #146, @robtaylor is planning to submit a patch for centos8 migration. Perhaps you should co-ordinate your efforts in that regard.

robtaylor, 2021-01-10:
@nadeemyaseen-lm could you share if you have anything working based on ubuntu?

mahmoodulhassan-lm, 2021-01-15:
Hi @robtaylor ,
I have built OpenLANE image based on Ubuntu OS. Can you please guide me on how to push my branch because I am facing an authentication error at my end?

agorararmard, 2021-01-15:
@mahmoodulhassan-lm: You need to create your own fork of OpenLANE, push your changes there (basing them on the develop branch), and then submit a PR. Then, one of the OpenLANE developers will review your changes and decides where to go from there (+ CI will auto-test your code).

https://github.com/efabless/openlane/blob/master/CONTRIBUTING.md

donn, 2021-06-16:
We're comfortable with CentOS 7 for now. Rocky Linux may be considered in the future as CentOS 7's EOL approaches.

closed 2021-06-16 by donn