The-OpenROAD-Project/OpenLane #17: Error during the Synthesis Step

harikumar27399, 2020-07-28:
I have tried using the OpenLane flow for the aardonyx SoC developed at IIT Madras, India. An error is encountered during the synthesis step using Yosys and ABC. 

The error message reads as below:
67.1. Executing OPT_EXPR pass (perform const folding).
terminate called after throwing an instance of 'std::out_of_range'
  what():  Cell::getParam()
[ERROR]: during executing: "yosys -c /openLANE_flow/scripts//synth.tcl -l /openLANE_flow/designs/aardonyx/runs/28-07_19-08//logs/synthesis/yosys.log |& tee >&@stdout"
[ERROR]: Last 10 lines:
child killed: SIGABRT

[error.log](https://github.com/efabless/openlane/files/4990837/error.log)
[yosys.log](https://github.com/efabless/openlane/files/4990840/yosys.log)

harikumar27399, 2020-07-30:
The verilog source folder for the SoC is: 
[soc.tar.gz](https://github.com/efabless/openlane/files/5001969/soc.tar.gz)

I kindly request you to look into the error.

ax3ghazy, 2020-07-31:
Hello @harikumar27399, 

I could replicate the yosys crash you're getting. Is there a way to configure the amount of memory in the SoC? If so, could you try reducing it and test again? Currently it has more than 4 KBytes of memory.

Thanks.

harikumar27399, 2020-08-06:
Thanks a lot @ax3ghazy. Configuring the memory helped cross the OPT_EXPR pass. However, now I am facing an issue in the later part of the synthesis. There is a particular cell '_DLATCH_P_' whose area remains unknown as it doesn't get assigned to any of the sky_pdk lib cells.
Due to this, the synthesis fails and the TCL interpreter exits with the following message:
ERROR: synthesis failed.
Exiting...

I feel this is also the reason behind slack violations pointed out by OpenSTA.

Updated verilog source directory: 
[aardonyx_src.tar.gz](https://github.com/efabless/openlane/files/5034384/aardonyx_src.tar.gz)

[opensta.log](https://github.com/efabless/openlane/files/5034386/opensta.log)
[yosys.log](https://github.com/efabless/openlane/files/5034390/yosys.log)

agorararmard, 2020-08-06:
@harikumar27399  As far as I understand, this means that the design contains latches and is not synthesizable by our version of Yosys. You should modify the design to remove those 6 latches. For example, this is a snippet of the Yosys log you attached reporting a latch in reg current_gate:
 ```Latch inferred for signal `$paramod\MakeClock\initVal=1'0\initGate=1'1.\current_gate' from process `$paramod\MakeClock\initVal=1'0\initGate=1'1.$proc$/openLANE_flow/designs/aardonyx/src/MakeClock.v:99$25234': $auto$proc_dlatch.cc:450:proc_dlatch$30639```

harikumar27399, 2020-08-07:
Thanks for the info @agorararmard. I removed those latches from the design and the synthesis passed successfully. However, there are timing violations (negative slack) reported after synthesis by OpenSTA. 

The libraries used for min. and max. STA calculations are:
LIB_MIN: sky130_fd_sc_hd__ss_100C_1v60.lib
LIB_MAX: sky130_fd_sc_hd__ff_100C_1v95.lib

OpenSTA report: wns = -0.52, tns = -1.62

The synthesis passes in spite of these violations. So, I would like to know whether I can proceed with the floorplan step by ignoring these violations. If not, is there a way to fix the negative slack by configuring the toolchain differently or by using a specific library for STA calculations?

[opensta.log](https://github.com/efabless/openlane/files/5041229/opensta.log)

agorararmard, 2020-08-07:
@harikumar27399 I'm afraid you're using the wrong corners for LIB_MIN and LIB_MAX, they should be:
``` 
::env(LIB_MIN) sky130_fd_sc_hd__ff_n40C_1v95.lib
::env(LIB_MAX) sky130_fd_sc_hd__ss_100C_1v60.lib
```
However, having a negative wns will not stop the flow, and it will proceed up to the GDS normally. We will be integrating some tools to enhance the wns as an optimization in the coming releases, however, for now, what you should do is increase the value of the environment variable ```CLOCK_PERIOD``` set inside config.tcl under your design until you no longer have a negative wns

harikumar27399, 2020-08-10:
@agorararmard I've solved the negative wns issue by increasing the clock period by one ns for the time being. I'm looking forward to the tools that will enhance the wns as an optimization in the coming releases.
Thank you.

closed 2020-09-04 by agorararmard