The-OpenROAD-Project/OpenLane #18: Iterations are keep going how long they will???

mdzakirhussain, 2020-07-29:
PROC:  Start NESTEROV's Optimization
PROC:    Global Lagrangian Multiplier is Applied

ITER: 0
    HPWL=382.550232
    OVFL=0.725638
    HPWL=(159.536499, 223.013733)
    POTN=2.085887E+08
    PHIC=1.524986E-09
    GRAD=4.210845E-01
    NuBT=0
    CPU =0.000000

ITER: 1
    HPWL=376.941772
    OVFL=0.697518
    HPWL=(154.447601, 222.494186)
    POTN=1.924570E+08
    PHIC=1.601235E-09

agorararmard, 2020-08-04:
@mdzakirhussain It takes a long time for large designs which could be considered to be a bug somehow in the placer. However, the iterations keep on going until OVFL is less than 0.1

agorararmard, 2020-09-28:
@mdzakirhussain, could you please try rc3 and report if the issue still stands?

agorararmard, 2020-10-01:
@mdzakirhussain, does this issue still stand or should I close it?

mdzakirhussain, 2020-10-01:
do not close, the problem still there allow me till 10/october/2020.

mdzakirhussain, 2020-10-12:
Now it is fine. Thank you.

closed 2020-10-12 by mdzakirhussain