The-OpenROAD-Project/OpenLane #19: Running ./flow.tcl gives Permission Denied error while creating folders

hadirkhan10, 2020-07-30:
Hi, I have cloned the repository in the aws Ubuntu instance in the following path `/home/merlproj/backend-tools`. I have built the skywater-pdk and installed it using the open_pdk. When I run the docker container from the path `/home/merlproj/backend-tools/openlane` the bash-4.1 shell comes up. I then run the script ./flow.tcl and it shows the following error:
[image]
I then tried to give the permission to flow.tcl file but the bash-4.1 does not recognise sudo command.

agorararmard, 2020-07-30:
@hadirkhan10 when mounting the docker: 
```docker run -it -v $(pwd):/openLANE_flow -v $PDK_ROOT:$PDK_ROOT -e PDK_ROOT=$PDK_ROOT -u $(id -u $USER):$(id -g $USER) openlane:rc2``` you mount as the currently logged in user. So, if the user doesn't have the permission to create the directory (i.e. outside of the docker, if you tried to create that same ./runs/ folder and failed) then you wouldn't be able to do it inside docker as well. (if that's not the case then we'll need to look into it.)

However, a quick workaround is mounting as root: ```docker run -it -v $(pwd):/openLANE_flow -v $PDK_ROOT:$PDK_ROOT -e PDK_ROOT=$PDK_ROOT  openlane:rc2``` However, this will mean that all folders created inside of the docker container are only editable by root (sudo) outside of the container.

hadirkhan10, 2020-07-30:
@agorararmard I have root privileges outside the container. I can create directories using the sudo command. But inside the docker container I am getting these errors

hadirkhan10, 2020-07-30:
@agorararmard the workaround command you provided works thanks! But please look into it without running it through root

agorararmard, 2020-07-30:
@hadirkhan10 did you run the docker using ```sudo docker ....``` or created the openlane directory with sudo?

hadirkhan10, 2020-07-30:
I created the openlane directory with "git clone" using the sudo command and run docker with sudo as well since docker gives error of running daemon without sudo.

agorararmard, 2020-07-30:
@hadirkhan10, that's the issue then. If you run ```git clone``` again without ```sudo``` everything should run perfectly without the need for sudo or root access

hadirkhan10, 2020-07-30:
@agorararmard well the point is that I created the openlane directory using the "git clone" in the home level directory which needs root permissions to create folder. So I needed to use "sudo git clone <url>" so that git is able to create a folder in the home directory without giving the permission denied error.

agorararmard, 2020-07-30:
@hadirkhan10  Yes that's why since you cloned the repo with sudo git clone, you will always have to edit it/run the docker in it as root.

closed 2020-07-30 by agorararmard