The-OpenROAD-Project/OpenLane #20: Perform custom pin-placement during Floorplanning

njose939, 2020-07-31:
Hi,
I guess the configurations for io pin placement is picked by openlane from /path/to/openlane/configuration/floorplan.tcl. However, I'm not sure of the steps to perform custom pin-placement during Floorplanning. 
For instance, for design, say picorv32a, if I want my 'resetn' pin to be precisely in the left, lower-half section; what all modifications do I need to do.

ax3ghazy, 2020-07-31:
Hi @njose939,

This is currently not very possible. The only options you have for pin placement currently are random equidistant distribution of pins, or a placement that optimizes the HPWL (see the `FP_IO_MODE` configuration parameter). What you describe is a feature we wanted to implement for a while, so I will leave this open until it's available.

Thanks.

agorararmard, 2020-11-16:
@nickson-jose: You can set `set ::env(FP_PIN_ORDER_CFG) $::env(DESIGN_DIR)/pin_order.cfg`and add a pin_order.cfg under your design directory like this one: https://github.com/efabless/openlane/blob/master/designs/spm/pin_order.cfg

Haseeb-Baloch, 2020-11-26:
The example pin_order.cfg sets all 32 pins of the 'x' input to the north, is there anyway to divide these pins to 2 sides?

agorararmard, 2020-11-26:
@Haseeb-Baloch: this will serve as a better example for you: https://github.com/efabless/caravel/blob/master/openlane/user_project_wrapper/pin_order.cfg

Haseeb-Baloch, 2020-11-26:
thank you very much

Haseeb-Baloch, 2020-11-26:
Thank you @agorararmard this was exactly what I was looking for

closed 2020-12-01 by nickson-jose

hanm2019, 2023-03-26:
Hi, I have a question about  `pin_order.cfg` file. It seem that It can custom the pins in which sides. 

If the pin_order.cfg is :

```
#N
x
y
z
```

Does it means that there are three pin in the north and the relative position of the three pins is decided, such as `x y z`? or  the `z x y` is also possible?