The-OpenROAD-Project/OpenLane #21: Inconsistensy between Routing strategies

diecaptain, 2020-08-02:
Build Information : Docker Build on Ubuntu 20.04
PDK Information : sky130A, variant - sky130_fd_sc_hd, typical op condition - tt_025C_1v80

I have observed the following error on two of my designs and on spi controller (raven_spi) from Raven-SOC.

[image]
The flow exits after track assignment begins when using routing strategies 0 through 3. I have used the inputs from tritonRoute.param file and have successfully ran TritonRoute on those independently. All the designs were successfully routed on routing strategy 14. 

Screenshot of PDK_PDK_VARIANT_config.tcl used on raven_spi is attached below:
[image]

I am unsure where the problem is. Kindly look into it.

Thank you.

agorararmard, 2020-08-04:
@diecaptain which version of open_pdks are you using/ could you share the tracks.info file? it should be copied into the run directory under tmp.

diecaptain, 2020-08-04:
I am using the open_pdks 1.0 tag rc2. The tracks.info_copy file in tmp directory is the same as the tracks.info file generated by open_pdks. The copy file is as below:

[image]

Thank you.

agorararmard, 2020-08-04:
@diecaptain Can you share all the logs under the run directory for any of the runs or can you share the (design, configurations) for us to replicate the error?

diecaptain, 2020-08-04:
This is the whole directory of raven_spi that I ran. The code itself is from efabless's Raven SOC. The run directory has two runs, the old one is the successful run using Routing strategy 14. The new one is the failed run using Routing strategy 0 which i reran now. I cleaned the failed runs so I dont have their logs. It fails on routing strategies 0,1,2 and 3, but passes on 14.

[raven_spi.tar.gz](https://github.com/efabless/openlane/files/5024536/raven_spi.tar.gz)

agorararmard, 2020-08-05:
@diecaptain I was unable to replicate the error you're facing. I ran with ROUTING_STRATEGY 0 and 1, using the openlane::rc2, and got a DRC/LVS/Antenna clean raven_spi run. However, looking at the logs you attached, I'm familiar with the pin access issue that is flagged in your tritonRoute log, as far as I've experienced, this happens when you use incompatible versions of skywater-pdk (newer) and open_pdks (older). So please make sure you checkout to the commits specified in the readme. TR14 works around this issue which is quite interesting.

If you insist on using the latest version of skywater-pdk then you might want to check this [#23](https://github.com/efabless/openlane/commit/c480afca55fa59b40a17da9d71b163391abf0c00). It's still being tested tho.

diecaptain, 2020-08-05:
Okay, thank you for your time. If I want to roll back my skywater-pdk version, which version is compatible with open_pdks 1.0 rc2? I didnt pay head to the pin access issue because after a few errors, it logs that 100% pin access was achieved. Right now, I have moved to steadily use TR14. However, while testing with the examples provided in openlane/designs, some of them do work for all routing strategies. It is also quite interesting to see that some designs give a clean run even with incompatible versions of skywater-pdk and open_pdks, that this issue is not uniform. Could you please guide me towards any documentation available on the differences between routing strategies 0 through 3 and 14? It would be a massive help to understand issues intuitively and come up with ideas.

Thank you very much for your time once again.

agorararmard, 2020-08-06:
This is a documentation of all configuration parameters used in OpenLane: [configuration/README.md](https://github.com/efabless/openlane/blob/master/configuration/README.md).

However, in brief, from 0 to 3, this is a variant from TR13, and they are not really different "routing strategies", but different "routing optimization iterations". You can check the source code here to see the differences, which only show up in this function ```FlexDR::main()``` [src](https://github.com/agorararmard/TritonRoute/blob/newopenroadTri/src/dr/FlexDR.cpp).
As for routing strategy 14, that uses TR14, which is much more 0-DRC capable but it takes more time and consumes more memory, that's why we are keeping the TR13 variant as well.

You can try them all on your design using an exploration sweep as described in this [file](https://github.com/efabless/openlane/blob/master/regression_results/README.md)

closed 2020-08-09 by diecaptain