The-OpenROAD-Project/OpenLane #22: How can multiple users on a machine run docker container?

hadirkhan10, 2020-08-04:
Hello, we have a team of multiple users working on an Amazon Web Services (AWS) Ubuntu machine and we would like all the users to run the docker container and the complete openlane flow. For that purpose I placed the openlane directory in the `/home` directory of the machine so that everyone can access it. But now docker requires sudo privileges to run which is not what we need. Is there a way to set up the openlane flow so that multiple users can work with the docker container and synthesize the designs?

agorararmard, 2020-08-04:
@hadirkhan10  As far as I know, this could only be done if all users have edit access to the folder in which openlane lies, and if home directory requires sudo access then all users should have sudo access to use it. However, mounting the docker as root (omitting the -u part when mounting) should solve the issue of using the flow, but for extracting an editable version of the runs, they should be copied to a folder where the user has edit access, or a sudo user should edit the access permissions on the runs after they are created.

hadirkhan10, 2020-08-04:
@agorararmard oh that's a problem. How would multiple members in the team then work collaboratively? If every user needs to clone the openlane  repository into their own user directories, that would not be viable at all. If we put the openlane folder in a common place it would definitely need read/write permissions. If we provide users read/write permissions specifically for that folder only, still they won't be able to run docker since it requires root privileges.

agorararmard, 2020-08-04:
@hadirkhan10 If all users are granted read/write/execute permissions on openlane and pdks directories then there would be no need for running the docker as root using this command (no root privileges):
```
docker run -it -v $(pwd):/openLANE_flow -v $PDK_ROOT:$PDK_ROOT -e PDK_ROOT=$PDK_ROOT -u $(id -u $USER):$(id -g $USER) openlane:rc2
```

hadirkhan10, 2020-08-04:
Okay @agorararmard I'll try and report back with this.

hadirkhan10, 2020-08-08:
@agorararmard I found a solution reading the docker documentation. For running the docker daemon without sudo privileges you need to add the user into the `docker` group created automatically. Once the user is in the `docker` group he/she can run the `docker run` command without the `sudo` keyword

closed 2020-08-08 by hadirkhan10