The-OpenROAD-Project/OpenLane #24: Error during Routing using FastRoute

harikumar27399, 2020-08-10:
While trying to use Openlane flow for the Aardonyx SoC designed at IIT Madras, there is a crash encountered during the routing process using FastRoute. 

The error message reads as below:
terminate called after throwing an instance of 'std::bad_alloc'
what():  std::bad_alloc
[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts//openroad/or_route.tcl |& tee >&@stdout /openLANE_flow/designs/aardonyx/runs/10-08_13-50//logs/routing/fastroute.log"
[ERROR]: Last 10 lines:
child killed: SIGABRT

FastRoute Log:
[fastroute.log](https://github.com/efabless/openlane/files/5052236/fastroute.log)

Configuration details:
set ::env(CLOCK_PERIOD) 11
set ::env(ROUTING_STRATEGY) 0
set ::env(GLB_RT_ADJUSTMENT) 0
set ::env(GLB_RT_L1_ADJUSTMENT) 0.99
set ::env(GLB_RT_L2_ADJUSTMENT) 0
set ::env(GLB_RT_MINLAYER) 1
set ::env(GLB_RT_MAXLAYER) 6
set ::env(GLB_RT_UNIDIRECTIONAL) 1
set ::env(GLB_RT_ALLOW_CONGESTION) 0
set ::env(GLB_RT_OVERFLOW_ITERS) 50

SoC Verilog Source Directory:
[aardonyx_soc.tar.gz](https://github.com/efabless/openlane/files/5052264/aardonyx_soc.tar.gz)

Kindly look into the issue.

smunaut, 2020-08-10:
You're running out of RAM.

harikumar27399, 2020-08-10:
@smunaut Is there any workaround for this?

harikumar27399, 2020-08-11:
Thanks for the info @smunaut . I circumvented the crash by increasing the swap size of my Ubuntu system.

closed 2020-08-11 by harikumar27399