The-OpenROAD-Project/OpenLane #26: Magic Hangs

harikumar27399, 2020-08-11:
While trying to use Openlane flow for the Aardonyx SoC designed at IIT Madras, there is a crash encountered during the run of Magic. The Ubuntu terminal hangs while writing to the LEF file after about 45 minutes of the start of Magic run. My system has 6GB of RAM and 6GB of swap space on top of it but the issue persists. I would like to know the recommended system RAM requirements for running Magic and the subsequent DRC and LVS checks. I would also like to know if there is any way to circumvent the issue faced. Do let me know if I need to provide any further details.

[magic.log](https://github.com/efabless/openlane/files/5058322/magic.log)

Config Details:
set ::env(CLOCK_PERIOD) 11
set ::env(RUN_MAGIC) 1
set ::env(RUN_ROUTING_DETAILED) 1
set ::env(RUN_SIMPLE_CTS) 0
set ::env(CLOCK_TREE_SYNTH) 1

Verilog Source Directory:
[aardonyx_soc.tar.gz](https://github.com/efabless/openlane/files/5058333/aardonyx_soc.tar.gz)

smunaut, 2020-08-11:
Just to give you a comparaison point, running the picorv32a design from the example (which is _way_ smaller than your design)
used about 16G of RAM and took about 8h to complete. Using any swap instead of real ram would probably make it substantially slower because of the swap in/out ... (and all that swap might make the whole machine unresponsive).

So you're going to need to go buy a whole lot more RAM ...

ax3ghazy, 2020-08-11:
(posted already on slack, but will post it here as well):
@harikumar27399: Could you share the routed DEF you got (under `designs/aardonyx/runs/<tag>/results/routing` ) to make it easier to reproduce?

harikumar27399, 2020-08-12:
@ax3ghazy I have shared the routed DEF file on slack for your perusal.

donn, 2021-06-16:
Stale, closing.

closed 2021-06-16 by donn