The-OpenROAD-Project/OpenLane #27: Cannot install via docker method on branch `develop`

frznchckn, 2020-08-12:
I'm using Ubuntu (see below) and this system successful build `rc2` and ran the example design `spm`
```
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
```

The docker build fails on `vlogtoverilog`
```
~/git/openlane-develop/docker_build$ make build-vlogtoverilog
mkdir -p logs/docker
export DOCKER_BUILDKIT=1;  docker build --rm  -t vlogtoverilog docker/vlogtoverilog | tee logs/docker/vlogtoverilog.build.txt
[+] Building 15.3s (5/16)                                                                                                                                                       
 => [internal] load build definition from Dockerfile                                                                                                                       0.0s
 => => transferring dockerfile: 38B                                                                                                                                        0.0s
 => [internal] load .dockerignore                                                                                                                                          0.0s
 => => transferring context: 2B                                                                                                                                            0.0s
 => [internal] load metadata for docker.io/library/centos:centos6                                                                                                          0.0s
 => CACHED [1/13] FROM docker.io/library/centos:centos6                                                                                                                    0.0s
 => ERROR [2/13] RUN yum -y install centos-release-scl &&     yum -y install devtoolset-6 devtoolset-6-libatomic-devel                                                    15.2s
------                                                                                                                                                                          
 > [2/13] RUN yum -y install centos-release-scl &&     yum -y install devtoolset-6 devtoolset-6-libatomic-devel:                                                                
#5 0.415 Loaded plugins: fastestmirror, ovl                                                                                                                                     
#5 0.622 Setting up Install Process                                                                                                                                             
#5 12.38 Resolving Dependencies                                                                                                                                                 
#5 12.38 --> Running transaction check                                                                                                                                          
#5 12.38 ---> Package centos-release-scl.noarch 10:7-4.el6.centos will be installed
#5 12.38 --> Processing Dependency: centos-release-scl-rh for package: 10:centos-release-scl-7-4.el6.centos.noarch
#5 12.48 --> Running transaction check
#5 12.48 ---> Package centos-release-scl-rh.noarch 0:2-4.el6.centos will be installed
#5 12.49 --> Finished Dependency Resolution
#5 12.50 
#5 12.50 Dependencies Resolved
#5 12.50 
#5 12.50 ================================================================================
#5 12.50  Package                   Arch       Version                  Repository  Size
#5 12.50 ================================================================================
#5 12.50 Installing:
#5 12.50  centos-release-scl        noarch     10:7-4.el6.centos        extras      12 k
#5 12.50 Installing for dependencies:
#5 12.50  centos-release-scl-rh     noarch     2-4.el6.centos           extras      12 k
#5 12.50 
#5 12.50 Transaction Summary
#5 12.50 ================================================================================
#5 12.50 Install       2 Package(s)
#5 12.50 
#5 12.50 Total download size: 24 k
#5 12.50 Installed size: 39 k
#5 12.50 Downloading Packages:
#5 12.78 --------------------------------------------------------------------------------
#5 12.78 Total                                            85 kB/s |  24 kB     00:00     
#5 12.78 warning: rpmts_HdrFromFdno: Header V3 RSA/SHA1 Signature, key ID c105b9de: NOKEY
#5 12.79 Retrieving key from file:///etc/pki/rpm-gpg/RPM-GPG-KEY-CentOS-6
#5 12.81 Importing GPG key 0xC105B9DE:
#5 12.81  Userid : CentOS-6 Key (CentOS 6 Official Signing Key) <centos-6-key@centos.org>
#5 12.81  Package: centos-release-6-10.el6.centos.12.3.x86_64 (@CentOS/6.10)
#5 12.81  From   : /etc/pki/rpm-gpg/RPM-GPG-KEY-CentOS-6
#5 12.83 Running rpm_check_debug
#5 12.83 Running Transaction Test
#5 12.84 Transaction Test Succeeded
#5 12.84 Running Transaction
  Installing : centos-release-scl-rh-2-4.el6.centos.noarch                  1/2 
  Installing : 10:centos-release-scl-7-4.el6.centos.noarch                  2/2 
  Verifying  : centos-release-scl-rh-2-4.el6.centos.noarch                  1/2 
  Verifying  : 10:centos-release-scl-7-4.el6.centos.noarch                  2/2 
#5 13.05 
#5 13.05 Installed:
#5 13.05   centos-release-scl.noarch 10:7-4.el6.centos                                   
#5 13.05 
#5 13.05 Dependency Installed:
#5 13.05   centos-release-scl-rh.noarch 0:2-4.el6.centos                                 
#5 13.05 
#5 13.05 Complete!
#5 13.19 Loaded plugins: fastestmirror, ovl
#5 13.19 Setting up Install Process
#5 13.26 Determining fastest mirrors
#5 13.77  * base: centos.host-engine.com
#5 13.77  * centos-sclo-rh: mirror.steadfastnet.com
#5 13.77  * centos-sclo-sclo: repos-tx.psychz.net
#5 13.77  * extras: repo1.dal.innoscale.net
#5 13.77  * updates: centos.host-engine.com
#5 14.97 No package devtoolset-6 available.
#5 15.08 No package devtoolset-6-libatomic-devel available.
#5 15.14 Error: Nothing to do
------
failed to solve with frontend dockerfile.v0: failed to build LLB: executor failed running [/bin/sh -c yum -y install centos-release-scl &&     yum -y install devtoolset-6 devtoolset-6-libatomic-devel]: runc did not terminate sucessfully
```
Seems odd as I don't think the dockerfile for this tool has changed between `rc2` and the HEAD of `develop`, but I'm just getting used to this repo, so I could be mistaken.

agorararmard, 2020-08-12:
@frznchckn the vlogtoverilog is not used in the block level of openlane. This docker file is deprecated, and it will be replaced with a newer version/or deleted soon. So there is no need to run make-build-vlog2verilog.

Also, there is no need to build any of the tools at all. Since they exist in .tar.gz inside the tar folder. All you have to do once you clone is go inside the docker_build file and run ```make merge``` and the build will go smoothly. Or better in the openlane directory run ```make openlane```.

agorararmard, 2020-09-04:
@frznchckn check https://github.com/efabless/openlane/commit/3b542ca08efc30e247db1ca0d0ae84cfbe72f201

closed 2020-09-04 by agorararmard