The-OpenROAD-Project/OpenLane #28: error during placement stage

praharshapm, 2020-08-19:
Hi,
I am getting error on running run_placement. Replace is causing the issue. 
`[INFO] 80pCellArea = -nan 
[INFO] FillerInit: TotalFillerArea = 821.911865 
[INFO] FillerInit: NumFillerCells = -2147483648 
[INFO] FillerInit: FillerCellArea = -nan 
[INFO] FillerInit: FillerCellSize = (-nan, -nan) 
[ERROR]: during executing: "replace < /openLANE_flow/scripts//replace_gp.tcl |& tee >&@stdout /openLANE_flow/designs/AMUX2_3V_top/runs/19-08_07-19//logs/placement/replace.log"
 [ERROR]: Last 10 lines: child killed: segmentation violation`

After debugging, I understood that `rep place_cell_nesterov_place` is causing error which is present in replace_gp.tcl

Can someone please help?

riking, 2020-08-19:
Input files are here: https://skywater-pdk.slack.com/archives/C016H8WJMBR/p1597847656281500?thread_ts=1597817669.264400&cid=C016H8WJMBR

ax3ghazy, 2020-08-24:
I understand this has been resolved, right?

praharshapm, 2020-08-24:
Yes. I actually increased the instance count by adding more circuitry to the design and thus the issue got resolved.

closed 2020-09-04 by agorararmard