The-OpenROAD-Project/OpenLane #30: Problem installing PDK in MAC

bsrinath10, 2020-09-01:
Dear Sir,
I tried installing the OPENLANE in my Mac OS. 
I tried installing the SkyWater PDK . I am facing the following issue given in the attachment. 
Kindly help me for successful installation . 

with regards
Srinath 
[image]

agorararmard, 2020-09-01:
@bsrinath10 could you share the full list of commands that you used for cloning/building/exporting PDK_ROOT etc.

However, a quick workaround would be building the OpenLANE docker first. Then mounting the docker and using it to install skywater-pdk and open_pdks since it already has all the dependencies built inside. (however, this is not a safe way to build the pdk in general, and it will make *almost* unusable outside of the docker.)

Don't forget to export PDK_ROOT before mounting the docker.

bsrinath10, 2020-09-02:
Dear Amr A Gouhar,

Thank you for your email.
I had executed the following commands in MAC OS terminal.
1. mkdir pdk
2. export PDK_ROOT = <>
3. cd ..
4. git clone git@github.com:efabless/openlane --branch rc2
5. cd openlane/docker_build
6. make merge
7. cd ..
8. docker run -it -v $(pwd):/openLANE_flow -v $PDK_ROOT:$PDK_ROOT -e
PDK_ROOT=$PDK_ROOT -u $(id -u $USER):$(id -g $USER) openlane:rc2
9. ./flow.tcl -design spm

But now am getting different output (as seen below).

[image: Screenshot 2020-09-02 at 3.44.55 PM.png]

Please help me for successful installation.
It would be better if you give me step-by-step installation procedure.

with regards
Dr B Srinath

bsrinath10, 2020-09-02:
Dear Amr A Gouhar, 

Thank you for your email. 
I had executed the following commands in MAC OS terminal. 
1. mkdir pdk 
2. export PDK_ROOT = <>
3. cd ..
4. git clone git@github.com:efabless/openlane --branch rc2
5. cd openlane/docker_build
6. make merge
7. cd ..
8. docker run -it -v $(pwd):/openLANE_flow -v $PDK_ROOT:$PDK_ROOT -e PDK_ROOT=$PDK_ROOT -u $(id -u $USER):$(id -g $USER) openlane:rc2
9. ./flow.tcl -design spm

But now am getting different output (as seen below). 
[image]

Please help me for successful installation. 
It would be better if you give me step-by-step installation procedure.

with regards 
Dr B Srinath

agorararmard, 2020-09-02:
@bsrinath10 it seems you have not installed the PDK. Please follow the following setup instructions in this order:
- https://github.com/efabless/openlane/#setting-up-the-pdk-skywater-pdk
- https://github.com/efabless/openlane/#setting-up-openlane

Which is summed up to running this sequence of commands:
Please make sure to export PDK_ROOT as a valid full path in your machine.
```bash
    export PDK_ROOT=<absolute path to where skywater-pdk and open_pdks will reside>
    cd  $PDK_ROOT
    git clone git@github.com:google/skywater-pdk.git
    cd skywater-pdk
    git checkout 4e5e318e0cc578090e1ae7d6f2cb1ec99f363120
    git submodule update --init libraries/sky130_fd_sc_hd/latest
    make sky130_fd_sc_hd 
    cd $PDK_ROOT
    git clone git@github.com:efabless/open_pdks.git -b rc2
    cd open_pdks
    make
    make install-local
```
Then do the openlane installation:
```bash
    git clone git@github.com:efabless/openlane --branch rc2
    cd openlane/docker_build
    make merge
    cd ..
```

agorararmard, 2020-09-03:
@bsrinath10 You may also want to look at [this](https://github.com/efabless/openlane/wiki#im-unable-to-clone-with-ssh), if you still have issues with cloning.

agorararmard, 2020-09-04:
@bsrinath10 Does the issue still stand?

bsrinath10, 2020-09-05:
Respected Sir,

Still the issue is not resolved .
Can you send me the link to install MAGIC .

yours sincerely
Dr Srinath

agorararmard, 2020-09-07:
@bsrinath10 http://opencircuitdesign.com/magic/index.html this is where you could get magic

bsrinath10, 2020-09-29:
Dear sir,
My issue still exists .
Also, I unable to install MAGIC layout. \

with regards
Dr B Srinath

agorararmard, 2020-09-29:
@bsrinath10 So if you pull master or rc3, you can do the following workaround to avoid installing magic.

```bash
export PDK_ROOT=<absolute path to where skywater-pdk and open_pdks will reside>
make skywater-pdk
make clone-open_pdks
docker run -it -v $(pwd):/openLANE_flow -v $PDK_ROOT:$PDK_ROOT -e PDK_ROOT=$PDK_ROOT -u $(id -u $USER):$(id -g $USER) openlane:rc3  bash -c "make install-open_pdks"
```

If the issue with conda still persists while running `make skywater-pdk` still persists, it would be better if you file an issue [here](https://github.com/google/skywater-pdk)

agorararmard, 2020-10-08:
@bsrinath10, did the provided workaround solve the issue?

d-m-bailey, 2020-11-08:
@bsrinath10 I wrote a summary about what I did to run openlane on macos. You can see it at https://skywater-pdk.slack.com/archives/C016UL7AQ73/p1604754202059400?thread_ts=1604715444.054100&cid=C016UL7AQ73
Does that help?

bsrinath10, 2020-11-09:
Dear Mitch Bailey,
I cannot access this slack.

Please help...

with regards
Srinath

d-m-bailey, 2020-11-09:
Does https://invite.skywater.tools/ work?

bsrinath10, 2020-11-09:
Dear Mitch Bailey,
Thank you.
But I am getting this error.
Please check the attachment

with regards
Dr Srinath

bsrinath10, 2020-11-10:
Dear Mitch Bailey,
Thank you very much for your email.
I got the procedure.

But am not using MAGIC layout. Am using KL layout.
I am interested to know how to see the placed standard cells in the layout..

d-m-bailey, 2020-11-10:
@bsrinath10 This is working now, right? Can we close this issue?

bsrinath10, 2020-11-10:
Dear Mitch Bailey,
How should I open the design after some time. ?
Is it mandatory that I have to do this  "export PDK_ROOT=pdks locations "
every time ?

with regards
Srinath

agorararmard, 2020-11-10:
@bsrinath10: You can use Klayout to view the design. You'll find the GDS-II under `runs/<run-tag>/results/magic/<design-name>.gds`.

Yes, you need to `export PDK_ROOT` every time you reboot your machine.

agorararmard, 2020-11-16:
@bsrinath10: I will assume this is resolved and close the issue. Please, reopen if proven otherwise.

closed 2020-11-16 by agorararmard