The-OpenROAD-Project/OpenLane #31: openLANE/efabless

chandni0760, 2020-09-03:
Hello I am Unable to clone the skywater-pdk  using the following command;
**git clone https://github.com/google/skywater-pdk**

PaulSchulz, 2020-09-03:
Work for me. What error do you get?

PaulSchulz, 2020-09-03:
What type of system are you using?

chandni0760, 2020-09-03:
Hi I am using Linux OS

PaulSchulz, 2020-09-03:
I'm using Ubuntu 20.04  (linux) and the command worked for me. What error do you get?

PaulSchulz, 2020-09-03:
You should see something like..

paul@silver:~/tmp$ git clone https://github.com/google/skywater-pdk
Cloning into 'skywater-pdk'...
remote: Enumerating objects: 26, done.
remote: Counting objects: 100% (26/26), done.
remote: Compressing objects: 100% (20/20), done.
remote: Total 903 (delta 11), reused 15 (delta 6), pack-reused 877
Receiving objects: 100% (903/903), 2.05 MiB | 1.62 MiB/s, done.
Resolving deltas: 100% (401/401), done.

chandni0760, 2020-09-03:
Hello Sir,

Can you tell me how much time does it take to install /clone the
skywater-pdk and nothing of this sort is coming as you have shown .
I had kept the run yesterday night and still its not done so asked for the
help
Will share you the screenshot of the same

PFA

PaulSchulz, 2020-09-03:
Command is type in at the terminal prompt.
- On my internet connection it takes less than 1 minute.

Can you 'ping' 'google.com' ?

chandni0760, 2020-09-03:
Sure let me try

Thanks

agorararmard, 2020-09-04:
@chandni0760 Does the issue still stand?

chandni0760, 2020-09-05:
Hi,

Yes May be I have issue in my Laptop itself can't install it on my laptop I
have also pinged Google.com and it works but issue lies in my system hence
I have already confronted the query to My IT Executive

Thanks a lot for the support!!

agorararmard, 2020-09-28:
@chandni0760, does the issue still stand?

agorararmard, 2020-11-16:
@chandni0760: I will assume this is resolved and close the issue. Please, reopen if it still stands.

closed 2020-11-16 by agorararmard