The-OpenROAD-Project/OpenLane #33: Overlap check failed during diode auto-placement

kdp1965, 2020-09-04:
Hi All,

I am working on a 16-Bit processor I wrote many years ago (named Pico16) using the OpenLane flow.  I am using the Docker image (downloaded last week) and have been able to get it all the way through the flow several times.  I have also been able to get a couple of the examples to run.  

I am making a few extensions to my Pico16 processor ISA, and occasionally when I run "flow.tcl -design pico16",  I get the following error (followed by some python error messages):

    Warning: Overlap check failed (2).
     PHY_470 overlaps PHY_1
     PHY_4877 overlaps PHY_429
    [ERROR]: during executing: "openroad -exit /openLANE_flow/scripts//openroad/or_diodes.tcl |& tee >&@stdout /openLANE_flow/designs/pico16/runs/04-09_19-06//logs//placement/diodes.log"
    [ERROR]: Last 10 lines:
    child process exited abnormally

    [ERROR]: Please check openroad  log file
    [ERROR]: Dumping to /openLANE_flow/designs/pico16/runs/04-09_19-06//error.log

Is this a known issue with a possible workaround?

Thanks,
Ken

agorararmard, 2020-09-04:
Hi @kdp1965, we're well aware of this issue and it has been solved in the develop and staging branches, and the fix will be coming to master very soon. However, the workaround for this in rc2 (master) is to reduce both PL_TARGET_DENSITY and FP_CORE_UTIL so that this "fake" overlap doesn't happen. To understand more about those configurations and what to do in general, [this](https://github.com/efabless/openlane/wiki#what-to-do-if-the-flow-failed-in-any-of-the-steps-from-floorplan-up-to-routing) would be a good place to start. You can use the exploration scripts, or explore manually since it's only one design.

This issue is a duplicate of #4, so, I'll close it.

closed 2020-09-04 by agorararmard

kdp1965, 2020-09-04:
Hi Amr,

Ahh, I saw the title of issue #4 but guess I didn't look at the body 
text not realizing it was the same issue.  Sorry for the duplicate.

I had read about the PL_TARGET_DENSITY some last night and wondered if 
this might be a way to work around it.  I'll try experimenting with 
that.  Thanks!

I have a general question:  Is there an openlane mailing list or user 
community?  Or is there just the Skywater PDK mailing list?  I don't see 
much activity on the Skywater PDK announcement mailing list.  I have 
several questions such as:

1.  In my openlane runs, I see many warnings (cell_name contains 0 
device instances) and don't really know if this is normal?
2.  Is there any work going on for a Register File compiler, or should I 
start such a project?
3.  How can I help add Yosys support for full-adder / half-adder cells?
4.  Is there a plan in the openlane flow for top level placement of 
things like OpenRAM cells, etc?
5.  etc., etc.

But I don't really want to start throwing questions like this out if 
it's the wrong forum.  Or perhaps some of these questions would simply 
be premature?

Thanks,
Ken

agorararmard, 2020-09-04:
Hi @kdp1965 , I guess the answer to all your questions is Join the community on [slack](https://invite.skywater.tools/)!