The-OpenROAD-Project/OpenLane #34: How to determine die area from successful flow?

SweeperAA, 2020-09-06:
I have successfully executed the top-level flow.tcl in the docker on one of the example designs (chacha in this case). Are there any results that provide the final die area required for the design (and what units the measurements are in)?

Thank you for the help!

agorararmard, 2020-09-07:
@SweeperAA, the die area is found inside the resulting def file. in designs/chacha/results/routing/chacha.def, divide by 10e6 and you'll get the die area by mm. However, if you want to generate a summary report for all the important statistics, you could use `run_designs.py` to run the design.
in the case of chacha
```
python3 run_designs.py -d chacha -t chacha_run
```
You'll find the generated summary under the regression_results folder. More about that in [here](https://github.com/efabless/openlane/blob/master/regression_results/README.md).

closed 2020-09-07 by agorararmard

SweeperAA, 2020-09-07:
Can you confirm that for:

`DIEAREA ( 0 0 ) ( 1077990 1088710 ) ;`

The die area would be approximately 1.17mm^2. That is, ( (1077990/1000000) * (1088710/1000000) ) = ~1.17mm^2

agorararmard, 2020-09-07:
@SweeperAA yes, that is correct.