The-OpenROAD-Project/OpenLane #35: Changing design configurations on the fly

nickson-jose, 2020-09-10:
Hi,

I want to know if openlane supports on-the-fly configuration changes. 
For instance: for a design, say picorv32a, I chose my `FP_CORE_UTIL` as 50 (in ~/picorv32a/config.tcl) ..I do till **run_floorplan** stage (in interactive flow) but now I want to change `FP_CORE_UTIL` to 60. So in this case, post modifying config.tcl,  do I need to run from the beginning of flow or just running **run_floorplan** again take `FP_CORE_UTIL` as 60 ?
Because the latter isn't updating my config.tcl (~/openlane/designs/runs/_date_/config.tcl).

agorararmard, 2020-09-10:
@nickson-jose I guess the best/safest way to do that is setting the variable again with the new value in your interactive script. i.e.:
```tcl
set ::env(FP_CORE_UTIL) 60
run_floorplan
```
and if you want the config to be stored under the config.tcl of that run, use `set_log` instead of `set`. Alternatively, you can 're-source' your config file using the `source_config <config_file>` command in your interactive script, if you want the change to come from a config file.

Also, if you change `FP_CORE_UTIL` value after running `run_floorplan` intuitively the change will not be reflected on the layout of the design unless you run `run_floorplan` again.

nickson-jose, 2020-09-10:
Hi @agorararmard .. tried the suggested with another switch `FP_IO_VMETAL` ..I initially ran with `FP_IO_VMETAL` 2..then modified `FP_IO_VMETAL` to 4...post modification ran only **run_floorplan**..the modifications weren't reflected in the layout as the vertical pins were still as per `FP_IO_VMETAL` 2

agorararmard, 2020-09-10:
@nickson-jose can you share the interactive script in which you did so? or the list of commands?

nickson-jose, 2020-09-11:
@agorararmard  Below are the configurations I changed during the flow (after first **run_floorplan**) which aren't getting reflected when i do **run_floorplan** again. However, these configurations are taken into consideration when I build from beginning again (i.e, from design -prep picorv32a). Below is the code snippet from `~/openlane/designs/picorv32a/config.tcl`

`set ::env(FP_CORE_UTIL)` 65
`set ::env(FP_IO_VMETAL)` 4
`set ::env(FP_IO_HMETAL)` 3

set filename $::env(OPENLANE_ROOT)/designs/$::env(DESIGN_NAME)/$::env(PDK)_$::env(PDK_VARIANT)_config.tcl
if { [file exists $filename] == 1} {
        source $filename
}

agorararmard, 2020-09-11:
@nickson-jose I'm talking about the interactive script, not the config.tcl. The config.tcl is only sourced once by default. To change that you need to do as I said above:
- Either set that config in the interactive script itself (the one you feed to the flow when using -interactive flag) using `set` and `set_log` 
- Or change the config.tcl and explicitly source it again in your interactive script using the command `source_config <config_file>`

So, can you share the interactive script in which you did so or the list of commands you use?

nickson-jose, 2020-09-11:
My bad @agorararmard ..below is the configuration i had set in the interactive script:

VirtualBox:~/Desktop/work/tools/openlane_working_dir/openlane$ **cat interactive_script** 
`set_log ::env(FP_ASPECT_RATIO)` 0.8

Am I'm sourcing it by:
bash-4.1$` ./flow.tcl -interactive -f interactive_script`

agorararmard, 2020-09-11:
@nickson-jose You should prep the design first, using the `prep` command, i.e.:
```tcl
prep -design <design_name if it's under designs or absolute path if it's not> -tag <the tag you used in the previous run that you want to continue on now> 
```
Also `interactive_script` should be `interactive_script.tcl`
So, it would be something like this:
```tcl
prep -design picorv32a -tag <the tag you used in the previous run that you want to continue on now> 
set ::env(FP_ASPECT_RATIO) 0.8
set_log ::env(FP_ASPECT_RATIO) 0.8
run_floorplan
```

nickson-jose, 2020-09-12:
@agorararmard ..got it working..thanks!

closed 2020-09-12 by nickson-jose