The-OpenROAD-Project/OpenLane #39: Segfault in TritonRoute

growly, 2020-09-23:
(Should this be filed with OpenROAD instead?)

I configured OpenLane and ran ./flow.tcl -design <my design> in the docker container. I'm running the rc2 release.

TritonRoute segfaulted and the whole flow died. I'm attaching the entire run output directory.

Please let me know if there's a way I can enable or provide more useful debug information.

[run.tar.gz](https://github.com/efabless/openlane/files/5271860/run.tar.gz)

agorararmard, 2020-09-24:
@growly, can you share with me the configuration files you fed to flow.tcl? I'm aware you included the config.tcl that gets generated inside the run directory. But, just to save time and allow me to replicate what happened to you, could you share the one you used?

Also, as a start, I'd advise using TR14 by setting ROUTING_STRATEGY to 14.

growly, 2020-09-24:
@agorararmard No problem, I think it's this one:

[config.tcl.gz](https://github.com/efabless/openlane/files/5278106/config.tcl.gz)

(This is just `gzip`ped, not `tar`'d)

ax3ghazy, 2020-09-26:
@growly: As per our discussion yesterday, I have run your design (on develop - rc3-to-be) with a resized synthesized memory (~25% of the current one) currently with 30% core utilization, 0.2 target placement density, 30ns clock period, and it got all the way to a GDSII with ~2 violations (that's where exploration comes in as discussed). I am doing another run with 50% synthesized memory on the side, but I expect that it will still take a while. 

Have you had a chance to re-run your design with TR14 and/or resizing the synthesized memory? From my observations, running your design as it is would require at least 16 GB of memory just during synthesis, and depending on the CPU and the design configuration, it is expected to take around 16-18 hours.

agorararmard, 2020-09-28:
@growly could you try rc3 and tell us if you have any trouble?

growly, 2020-09-29:
Thanks for the updates, I'm working on reproducing your results!

growly, 2020-10-08:
I have been running the new rc3 flow for about the past week. It definitely seems more stable. I certainly haven't run into any segfaults this time.

agorararmard, 2020-10-08:
@growly: That's great to hear! 

Would you mind sharing your design in the repo (this way, we'll always be sure that it's able to pass the flow)?

You can do that by submitting a PR to the [develop branch](https://github.com/efabless/openlane/tree/develop) in which you put your design and configuration under `./designs/`. Make sure to include a LICENSE accrediting yourself as the developer of this Verilog code. Of, course we'll understand if you don't want to share it on the repo, so no pressure.

Also, I'll close this issue since it no longer stands.

closed 2020-10-08 by agorararmard