The-OpenROAD-Project/OpenLane #40: Latches not mapped by yosys to skywater-130 cells

growly, 2020-09-23:
Running the OpenLane flow on the trivial latch

```
module test_latch (
  input wr_en,
  input a,
  output b
);
​
  reg a_latch;
​
  always @(*) begin
    if (wr_en)
      a_latch = a;
  end
​
  assign b = a_latch;
```

fails synthesis. Yosys correctly infers a `$_DLATCH_P_` primitive but cannot map this to a suitable cell, like Skywater 130's `dlxtp` cell. Cadence Genus will do this:

```​
module test_latch(a, wr_en, b);
  input a, wr_en;
  output b;
  wire a, wr_en;
  wire b;
  wire n_0, n_2;
  sky130_fd_sc_hd__inv_8 g4(.A (n_0), .Y (b));
  sky130_fd_sc_hd__inv_1 g5(.A (n_2), .Y (n_0));
  sky130_fd_sc_hd__dlxtp_1 a_latch_reg(.GATE (wr_en), .D (a), .Q (n_2));
endmodule
```

Yosys seems to need special tech-mapping for this primitive, which is provided by (for example) FPGA targets, but not for Skywater.

I created a test mapping /openLANE_flow/dlatchp.v

```
module \$_DLATCH_P_ (input E, input D, output Q);
  sky130_fd_sc_hd__dlxtp_1 _TECHMAP_REPLACE_ (
    .GATE(e),
    .D(D),
    .Q(Q));
endmodule
```

and inserted `techmap -map /openLANE_flow/dlatchp.v` into `scripts/synth.tcl` to make sure the mapping was run. This seems to correct the issue and synthesis passes. The flows then continues before segfaulting on the `replace` command. So maybe I did it wrong.

smunaut, 2020-09-24:
Lots of the latches seems to have been excluded from mapping in `no_synth.cells` ( in open_pdk ), you can try removing them from the excluded list and see if that works.
(my understanding is they blacklisted a bunch of stuff they just hadn't tested yet ...)

growly, 2020-09-24:
I tried removing the latches from the no_synth.cells list, but they still don't get mapped. As far as I can tell, no_synth.cells is used by `scripts/libtrim.pl` to generated a `trimmed.lib`, but this is only used by the yosys function `dfflibmap`. Latches are "simple" cells which yosys seems to map only through `techmap` (or maybe `simplemap`?)

riking, 2020-09-26:
> [... failing] on the replace command. So maybe I did it wrong.

You have one uppercase and one lowercase E - is that a problem?

growly, 2020-09-26:
> > [... failing] on the replace command. So maybe I did it wrong.
> 
> You have one uppercase and one lowercase E - is that a problem?

Good catch, but that's just a typo from when I copied the text into the bug. Synthesis passes with the techmap passed in (so the mapping works).

agorararmard, 2020-09-28:
@growly, replace should now give you more indicative error messages on rc3. So, please try rerunning it. Also, An example configuration of a tiny design that doesn't have a clock looks like this:
```tcl
set ::env(RUN_SIMPLE_CTS) 0
set ::env(CLOCK_TREE_SYNTH) 0
set ::env(FP_SIZING) absolute
set ::env(DIE_AREA) "0 0 160 160"
set ::env(PL_TARGET_DENSITY) 0.20
set ::env(SYNTH_MAX_FANOUT) 6
set ::env(PL_SKIP_INITIAL_PLACEMENT) 1
```
Give it a try and also check [this](https://github.com/efabless/openlane/wiki#how-to-add-a-small-design) if you still have any problems.

agorararmard, 2020-10-08:
@growly, any luck with this?

agorararmard, 2020-11-16:
@growly: Since this has been incorporated into openlane and open_pdks, I'll assume this issue is fixed. Please, reopen, if you found otherwise.

closed 2020-11-16 by agorararmard

growly, 2020-11-18:
Sorry this fell off my radar. My issue is fixed!