The-OpenROAD-Project/OpenLane #42: why I am getting an error after performing "make test"

mdzakirhussain, 2020-09-28:
make test
cd /home/zakir/openlane && \
	docker run -it -v /home/zakir/openlane:/openLANE_flow -v /home/zakir/opl/:/home/zakir/opl/ -e PDK_ROOT=/home/zakir/opl/ -u 0:0 openlane:rc3 bash -c "./flow.tcl -design spm -tag openlane_test -overwrite"
[INFO]: 
      ___   ____   ___  ____   _       ____  ____     ___
     /   \ |    \ /  _]|    \ | |     /    ||    \   /  _]
    |     ||  o  )  [_ |  _  || |    |  o  ||  _  | /  [_
    |  O  ||   _/    _]|  |  || |___ |     ||  |  ||    _]
    |     ||  | |   [_ |  |  ||     ||  _  ||  |  ||   [_
     \___/ |__| |_____||__|__||_____||__|__||__|__||_____|

[INFO]: Version: rc3
[INFO]: Running non-interactively
[INFO]: Using design configuration at /openLANE_flow/designs/spm/config.tcl
[INFO]: PDKs root directory: /home/zakir/opl/
[INFO]: PDK: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
couldn't read file "/home/zakir/opl//sky130A/libs.tech/openlane/config.tcl": no such file or directory
    while executing
"source $pdk_config"
    (procedure "prep" line 123)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 9)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    puts_info "Running interactively"
    if { [info exists arg_values(..."
    (file "./flow.tcl" line 160)
Makefile:86: recipe for target 'test' failed
make: *** [test] Error 1

agorararmard, 2020-09-28:
@mdzakirhussain from the attached log, I see:
```
couldn't read file "/home/zakir/opl//sky130A/libs.tech/openlane/config.tcl": no such file or directory
while executing
```
Which means you haven't installed (properly) the pdk. Did you run `make` first or at least `make pdk`?

mdzakirhussain, 2020-09-28:
I just followed the steps given in Quick Start:

export PDK_ROOT=<absolute path to where skywater-pdk and open_pdks will reside>
    make
    make test # This is to test that the flow and the pdk were properly installed

agorararmard, 2020-09-28:
@mdzakirhussain Can you share the log of `make`?
Also, make sure that you have the magic prerequisite installed as per [this](https://github.com/efabless/openlane#prerequisites), check the version number.
It might be a disk space issue, so check that as well.

mdzakirhussain, 2020-09-28:
Ok. Could you please answer the following 
1. which operating system is used for this tool
2. What is the disk space required 
3. What is the minimum RAM Required
4. Should we have to install tools as root or normal user

agorararmard, 2020-09-29:
@mdzakirhussain :
1. I've used it on different versions of Linux without any trouble, but it's supposed to be usable on Mac as well. Here is the link to the website: http://opencircuitdesign.com/magic/index.html
2. Roughly ~2GB. (This is for the pdk, not magic)
3. I guess any size of RAM would do the job, but it will be a limiting factor on the size of the designs you run.
4. It's your call. Both should work just as fine.

mdzakirhussain, 2020-10-10:
@agorararmard :

I am unable to run new designs, kindly check the screenshots to see what error I am getting

[image]
[image]
[image]

mdzakirhussain, 2020-10-12:
Thank you I resolved this.

closed 2020-10-12 by mdzakirhussain

andrsmllr, 2021-03-14:
@mdzakirhussain I am seeing a similar issue. Could you describe which steps resolved the issue in your case?

mdzakirhussain, 2021-03-16:
@andrsmllr, your design folder must reside inside the "designs" folder which is inside the openlane_master directory. Of course, you should be following the instructions given at https://github.com/efabless/openlane/blob/master/designs/README.md