The-OpenROAD-Project/OpenLane #67: How to modify config.tcl for simple combinational circuits? In which there in no clock.

mdzakirhussain, 2020-10-12:
Let's take the following example 

module fa(a,b,c,sum,co);
input a,b,c;
output sum,co;
assign sum=a^b^c;
assign co=a&b|a&c|b&c;
endmodule

In the above example there is no clock, how do I modify the config.tcl

I tried commenting the following it is give error, and by keeping that as well the same issue. Please help.

# Fill this
set ::env(CLOCK_PERIOD) "10"
set ::env(CLOCK_PORT) "clk"
set ::env(CLOCK_NET) $::env(CLOCK_PORT)

agorararmard, 2020-10-13:
@mdzakirhussain, you should simply add:

```
set ::env(CLOCK_TREE_SYNTH) 0
```
To skip cts, since there is no clock.

mdzakirhussain, 2020-10-14:
@agorararmard, Kindly check this log report, I am getting this error [ERROR] Filler area is negative!!

After setting set ::env(CLOCK_TREE_SYNTH) 0
---------------------------------------------------------

OpenROAD 0.9.0 bbd0ededbc
This program is licensed under the BSD-3 license. See the LICENSE file for details. 
Components of this program may be licensed under more restrictive licenses which must be honored.
Notice 0: Reading LEF file:  /openLANE_flow/openlane/designs/fa/runs/fa2/tmp/merged_unpadded.lef
Notice 0:     Created 11 technology layers
Notice 0:     Created 25 technology vias
Notice 0:     Created 437 library cells
Notice 0: Finished LEF file:  /openLANE_flow/openlane/designs/fa/runs/fa2/tmp/merged_unpadded.lef
Notice 0: 
Reading DEF file: /openLANE_flow/openlane/designs/fa/runs/fa2/results/floorplan/fa.floorplan.def
Notice 0: Design: fa
Notice 0:     Created 5 pins.
Notice 0:     Created 14 components and 42 component-terminals.
Notice 0:     Created 7 nets and 14 connections.
Notice 0: Finished DEF file: /openLANE_flow/openlane/designs/fa/runs/fa2/results/floorplan/fa.floorplan.def
[INFO] DBU = 1000
[INFO] SiteSize = (460, 2720)
[INFO] CoreAreaLxLy = (5520, 10880)
[INFO] CoreAreaUxUy = (20700, 24480)
[INFO] NumInstances = 14
[INFO] NumPlaceInstances = 4
[INFO] NumFixedInstances = 10
[INFO] NumDummyInstances = 0
[INFO] NumNets = 7
[INFO] NumPins = 19
[INFO] DieAreaLxLy = (0, 0)
[INFO] DieAreaUxUy = (26620, 37340)
[INFO] CoreAreaLxLy = (5520, 10880)
[INFO] CoreAreaUxUy = (20700, 24480)
[INFO] CoreArea = 206448000
[INFO] NonPlaceInstsArea = 37536000
[INFO] PlaceInstsArea = 81328000
[INFO] Util(%) = 48.148148
[INFO] StdInstsArea = 81328000
[INFO] MacroInstsArea = 0
[InitialPlace]  Iter: 1 CG Error: 4.81814e-09 HPWL: 200340
[InitialPlace]  Iter: 2 CG Error: 1.1724e-09 HPWL: 176098
[InitialPlace]  Iter: 3 CG Error: 7.99099e-10 HPWL: 176468
[InitialPlace]  Iter: 4 CG Error: 4.14999e-10 HPWL: 176672
[InitialPlace]  Iter: 5 CG Error: 1.35229e-10 HPWL: 176802
[ERROR] Filler area is negative!!
       Please put higher target density or 
       Re-floorplan to have enough coreArea
 (REPL-1)
Error: RePlAce terminated with errors.

agorararmard, 2020-10-14:
@mdzakirhussain: This is because the design is too small. Can you try appending these to your configuration file (config.tcl):
```tcl
set ::env(FP_CORE_UTIL) 5
set ::env(PL_TARGET_DENSITY) 0.5
```
(low utilization)

If these don't work, you may want to take a look at:
https://github.com/efabless/openlane/blob/master/regression_results/README.md
and explore different utilizations and densities.

agorararmard, 2020-10-28:
@mdzakirhussain: Did my previous answer solve this issue?

mdzakirhussain, 2020-10-28:
@agorararmard , kindly allow me one-day, due to other works I could not able to test. Sorry about that. For sure, tomorrow I will update.

Thank you.

mdzakirhussain, 2020-10-30:
@agorararmard , yes your solution perfectly worked Thank you.

Could you please help me understand on what basis we decide  values for the following

FP_CORE_UTIL & PL_TARGET_DENSITY

set ::env(FP_CORE_UTIL)  ???????
set ::env(PL_TARGET_DENSITY) ???????

agorararmard, 2020-10-30:
@mdzakirhussain: That's great to hear.

The basis is the complexity of the design. But, mostly, we obtain these values by running the exploration script on the design. You could find more about the meaning of these values [here](https://github.com/efabless/openlane/wiki#what-is-the-difference-between-fp_core_util-and-pl_target_density) and more about the exploration script [here](https://github.com/efabless/openlane/wiki#how-to-get-to-0-drc-violations). However, in your case the design was too small so it was a special case that you could know more about [here](https://github.com/efabless/openlane/wiki#how-to-add-a-small-design). I will updating this last one soon with better guidelines.

I'll close this issue since it's solved and we could resume the discussion on [slack](https://invite.skywater.tools/).

closed 2020-10-30 by agorararmard