The-OpenROAD-Project/OpenLane #78: rc3: OpenPhySyn Segmentation Fault

aunali1, 2020-10-20:
Fresh setup of PDKs and OpenLane. Using `sky130_fd_sc_hdll` library.

Ran the simple `./flow.tcl -design spm` and got the following:
```
[OpenPhySyn] [2020-10-20 07:05:51.834] [info] Invoking repair_timing transform
[OpenPhySyn] [2020-10-20 07:05:51.840] [info] Buffer library: sky130_fd_sc_hdll__buf_4, sky130_fd_sc_hdll__buf_8, sky130_fd_sc_hdll__buf_1
[OpenPhySyn] [2020-10-20 07:05:51.840] [info] Inverter library: None
[OpenPhySyn] [2020-10-20 07:05:51.840] [info] Buffering: enabled
[OpenPhySyn] [2020-10-20 07:05:51.840] [info] Driver sizing: enabled
[OpenPhySyn] [2020-10-20 07:05:51.840] [info] Pin-swapping: enabled
[OpenPhySyn] [2020-10-20 07:05:51.840] [info] Mode: Timing-Driven
[OpenPhySyn] [2020-10-20 07:05:51.840] [info] Iteration 1
[ERROR]: during executing: "Psn /openLANE_flow/scripts/openPhySyn.tcl |& tee >&@stdout /openLANE_flow/designs/spm/runs/20-10_07-05/logs/placement/openphysyn.log"
[ERROR]: Last 10 lines:
child killed: segmentation violation

[ERROR]: Please check Psn  log file
[ERROR]: Dumping to /openLANE_flow/designs/spm/runs/20-10_07-05/error.log

    while executing
"try_catch Psn $::env(SCRIPTS_DIR)/openPhySyn.tcl |& tee $::env(TERMINAL_OUTPUT) $::env(openphysyn_log_file_tag).log"
    (procedure "run_openPhySyn" line 7)
    invoked from within
"run_openPhySyn"
    (procedure "run_placement" line 13)
    invoked from within
"run_placement"
    (procedure "run_non_interactive_mode" line 13)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    puts_info "Running interactively"
    if { [info exists arg_values(..."
    (file "./flow.tcl" line 160)
```

`dmesg` reports the segfault as:
```
[15672.622994] Psn[38572]: segfault at 18 ip 00000000008962cf sp 00007fff5f687630 error 4 cpu 4 in Psn[400000+15d6000]
[15672.623000] Code: 89 ef ff 50 10 4c 89 ef 84 c0 49 8b 45 00 0f 84 e7 00 00 00 ff 50 18 48 8b 7b 20 49 89 c7 48 8b 07 4c 89 fe ff 90 58 02 00 00 <48> 8b 78 18 49 89 c6 e8 f5 2d fb ff 48 8b 7b 48 4c 89 fe 84 c0 75
```

If it helps, host distro is Clear Linux.

d-m-bailey, 2020-10-20:
May or may not be related, but some scripts are only setup for 2 character libraries, i.e. hd, hs, ls, ms.

For example, the netgen script in `open_pdks/sky130/netgen/sky130_setup.tcl` has the following lines which will only apply to 2 character libraries (note the \w\w):
```
#---------------------------------------------------------------
# Digital cells (ignore decap, fill, and tap cells)
# Make a separate list for each supported library
#---------------------------------------------------------------
# e.g., ignore class "-circuit2 sky130_fc_sc_hd_decap_3"
#---------------------------------------------------------------

foreach cell $cells1 {
    if {[regexp {sky130_fd_sc_\w\w__decap_[[:digit:]]+} $cell match]} {
        ignore class "-circuit1 $cell"
    }
    if {[regexp {sky130_fd_sc_\w\w__fill_[[:digit:]]+} $cell match]} {
        ignore class "-circuit1 $cell"
    }
}
foreach cell $cells2 {
    if {[regexp {sky130_fd_sc_\w\w__decap_[[:digit:]]+} $cell match]} {
        ignore class "-circuit2 $cell"
    }
    if {[regexp {sky130_fd_sc_\w\w__fill_[[:digit:]]+} $cell match]} {
        ignore class "-circuit2 $cell"
    }
}

```

agorararmard, 2020-10-20:
@aunali1, by `Fresh setup of PDKs`, do you mean building the latest pdk or using the commit hashes in our README.md? Could you share the commit hashes you used?

agorararmard, 2020-10-20:
@d-m-bailey, this is a great catch, but it's more related to the pdk setup on [open_pdks](https://github.com/RTimothyEdwards/open_pdks/), so it would be wonderful if you can open an issue [there](https://github.com/RTimothyEdwards/open_pdks/issues) or fix it locally and submit a [PR](https://github.com/RTimothyEdwards/open_pdks/pulls). We'll defenitly appreciate that!

aunali1, 2020-10-20:
@agorararmard Sorry, should have clarified. Yes, I am using `open_pdks` HEAD at commit [`859ff4b`](https://github.com/RTimothyEdwards/open_pdks/commit/859ff4b1b84a550fbecf7e7bf0870c1c181fb0c9) and `skywater-pdk` HEAD at commit [`fb266d1`](https://github.com/google/skywater-pdk/commit/fb266d108161a506fc681d48eeb69bfbc34d41f6). I have also used the default `sky130_fd_sc_hd` library with the same results.

agorararmard, 2020-10-20:
@aunali1: That's one of the reasons we have specific commit hashes in the README and the Makefile. The latest version of the PDK is broken for our purposes, but it will be updated soon with the fixes. Check issue [172](https://github.com/google/skywater-pdk/issues/172). Please, use the exact commit hashes that we specify or just export `PDK_ROOT` and run `make pdk`, otherwise we cannot guarantee any outcome. 
This is not an OpenLANE issue so I'm closing it.

closed 2020-10-20 by agorararmard

mithro, 2020-10-20:
@agorararmard -- FWIW The tools should never segfault no matter what input it is given (even if the input is garbage) -- it should return an error message describing what the problem is (and hopefully what needs to be changed to make the problem go away).