The-OpenROAD-Project/OpenLane #79: SYNTHESIS ERROR: NO NETLIST BEING GENERATED

wajehulhasan, 2020-10-25:
Hello, I was running the design with the memory module set as a black_box as mentioned in `https://github.com/efabless/openlane/blob/master/doc/chip_integration.md`  by adding the below two variables in the config.tcl script:
```
set ::env(VERILOG_FILES_BLACKBOX) $::env(OPENLANE_ROOT)/designs/memory_core/src/sram.v
set ::env(SYNTH_READ_BLACKBOX_LIB) 1
```
My design directory `src` contains the following three files:

- `memory_core.v` 
- `core.v` 
- `sram.v`

With `memory_core` being the top module.

However, after synthesis I realized that although there were no errors but no mapping was done and an empty netlist was generated which contained only these lines:
```
module memory_core(clk, reset);
          input clk;
          input reset;
endmodule
```

 I went through the log files and observed that it is not creating anything. It had the following the messages:
```
Executing ALUMACC pass (create $alu and $macc cells).
Extracting $alu and $macc cells in module memory_core:
created 0 $alu and 0 $macc cells
```

```
Creating internal representation of mux trees.
No muxes found in this module.
Removed 0 multiplexer ports.
```

```
Extracting gate netlist of module `\memory_core' to `/tmp/yosys-abc-BDpQ39/input.blif'..
Extracted 0 gates and 0 wires to a netlist network with 0 inputs and 0 outputs.
Don't call ABC as there is nothing to map
```

agorararmard, 2020-10-28:
@wajehulhasan: I think we'll need to look at the rest of the configs and the interactive script that you're using.

agorararmard, 2020-11-16:
@wajehulhasan: Does this issue still stand?

agorararmard, 2020-12-09:
@wajehulhasan: Any follow-up on this?

agorararmard, 2020-12-21:
@wajehulhasan: Given the current state of your project, I'll assume this issue has been resolved. Please re-open if you believe otherwise.

closed 2020-12-21 by agorararmard