The-OpenROAD-Project/OpenLane #82: `make test` failure on rc4

jeras, 2020-11-05:
I First tried openlane rc3 last week, I was able to install all dependencies and to add myself to the docker group.
I was able to run `make test` without issues. This is on Ubuntu 20.04.

Today I tried `rc4` twice, the second time with a thorough cleanup (not sure about docker).
`make test` ends with the next message:
```
[INFO]: /openLANE_flow/designs/spm/runs/openlane_test/results/magic/spm.spice against /openLANE_flow/designs/spm/runs/openlane_test/results/lvs/spm.lvs.powered.v
[INFO]: Running Antenna Checks...
[INFO]: Running OpenROAD Antenna Rule Checker...
[INFO]: Generating Final Summary Report...
[SUCCESS]: Flow Completed Without Fatal Errors.
/bin/sh: 1: [[: not found
Basic test failed
```

riking, 2020-11-06:
`/bin/sh: 1: [[: not found`

A script using bashisms is missing the `#!/bin/bash`.

ax3ghazy, 2020-11-06:
@jeras, @riking: Thanks! This is now fixed in develop.

agorararmard, 2020-11-13:
I'm re-opening this until the fix reaches master.

commit 48bdb4dd (2020-11-06): Get rid of the bashist [[ ]] (fixes #82)

- also add a missing shebang in travisBuild.sh
closed 2020-12-05 by ax3ghazy