The-OpenROAD-Project/OpenLane #84: missing connections in generated spice netlist

jeras, 2020-11-08:
Using rc4.
I have synthesized the `inverter` to try running a spice simulation.
The generated spice file was `designs/inverter/runs/07-11_21-11/results/magic/inverter.spice`.
I expected a full parasitic extraction netlist, but I am actually more happy with what I got, which is a netlist of standard cells (including capacitors filler and tap cells).

After I tried to create an example simulation and I got the next error.
```
Error: too few nodes: xfiller_0_15 vgnd vpwr sky130_fd_sc_hd__decap_12
```

The affected line in the spice netlist is:
```
XFILLER_0_15 VGND VPWR sky130_fd_sc_hd__decap_12
```

The problem seems two missing connections for ports `VNB` & `VPB` for a `decap` cell.
```
.subckt sky130_fd_sc_hd__decap_12 VGND VNB VPB VPWR
M1000 VPWR VGND VPWR VPB phighvt w=870000u l=4.73e+06u
+  ad=4.524e+11p pd=4.52e+06u as=0p ps=0u
M1001 VGND VPWR VGND VNB nshort w=550000u l=4.73e+06u
+  ad=2.86e+11p pd=3.24e+06u as=0p ps=0u
.ends
```
The same ports are also present on filler cells and also not connected in the netlist.
```
.subckt sky130_fd_sc_hd__fill_1 VGND VNB VPB VPWR
.ends
```

Interestingly, sub-circuit instances are also not named consistently.
`XFILLER_*_*` is used for `decap` and `fill` cells, some instances use the `XPHY_*` name.

d-m-bailey, 2020-11-08:
@jeras This may be a problem that has already been fixed. 

```
open_pdks: commit c2662b9
skywater-pdk: commit 9fcfb4c
```
gives the following with openlane rc4
`XFILLER_0_15 VGND VGND VPWR VPWR sky130_fd_sc_hd__decap_12
`

jeras, 2020-11-09:
Thanks @d-m-bailey,
I could not find an appropriate branch using this versions, so I just modified `Makefile` in the `rc4` version.
I was able to run `make` and to run `./flow.tcl -design inverter` inside the docker image.
The resulting `inverter.spice` file indeed has the all four connections to `decap` and `fill` cells.

Sub-circuit instance naming remains the same as before.

agorararmard, 2020-11-09:
@jeras: There are currently no active branches using the latest (open_pdks, skywater-pdk) version as we're waiting to settle for a solution for a fake DRC violation that shows up. If you go to `logs/magic/magic.drc`, you will find the following violation in all the designs: `All nwells must contain metal-connected N+ taps`. This happens because the abstract views are used in openlane's DRC, and in the LEF file the tap cells are connected by name. But, if you run DRC on the resulting GDS-II, you won't see this violation as the tap cells will be indeed connected.

One solution would be simply showing a warning that this specific violations might is fake if you already have tap cells in your design.

jeras, 2020-11-09:
Thanks @agorararmard for the explanation. For now I do not care about DRC, I am not there yet.

I was now able to get a few steps further running the spice simulation. I get the next error.
```
Scale set
Error on line 0 :
  m.x1.xfiller_0_15.x0.msky130_fd_pr__pfet_01v8_hvt vpwr 0 vpwr vpwr x1.xfiller_0_15.x0:sky130_fd_pr__pfet_01v8_hvt__model l=    4.72999999999999954e+00   w=    8.69999999999999996e-01   ad=    0.00000000000000000e+00   as=    0.00000000000000000e+00   pd=    0.00000000000000000e+00   ps=    0.00000000000000000e+00   nrd=    0.00000000000000000e+00   nrs=    0.00000000000000000e+00   sa=    0.00000000000000000e+00   sb=    0.00000000000000000e+00   sd=    0.00000000000000000e+00   nf=    1.00000000000000000e+00 
could not find a valid modelname
```
First I thought `ngspice` was unable to find some source files, so I experimented with the `NGSPICE_INPUT_DIR` system variable. But apparently if a file was not found I would get something like:
```
Error: Could not find include file sky130_fd_sc_hd.spice
```
So now I am not sure what is going on, and I do not have enough spice experience to debug this quickly.

jeras, 2020-11-09:
The last issue I described is separate, also reported here https://github.com/google/skywater-pdk/issues/228

I will let the maintainer decide whether to close this issue now or when the right PDK branch starts being used.

agorararmard, 2020-11-09:
@jeras: I'd rather keep it open until it's resolved in skywater-pdk as a reminder for us to keep tracking. Thank you for opening that discussion on slack! (but maybe rename the issue?)

agorararmard, 2020-12-07:
@jeras: The current version of skywater-pdk on master doesn't have this issue anymore (according to the issue you referenced above). Therefore, I'm closing this issue as it's now resolved. Please, reopen if you find otherwise.

closed 2020-12-07 by agorararmard