The-OpenROAD-Project/OpenLane #90: "diodes.def does not exist." when using (DIODE_INSERTION_STRATEGY = 2)

dan-rodrigues, 2020-11-15:
Adding this to the `spm` config.tcl file (plus others I've tried) causes errors such as [the one in this gist](https://gist.github.com/dan-rodrigues/2523374cece73471d7fd0e797c628f60):

`set ::env(DIODE_INSERTION_STRATEGY) 2`

The default value (3?) doesn't seem to have this issue.

The logs in the linked gist are from running in docker on `develop` (commit d03377b84212527a269dfc58bb637a9e921af150)

agorararmard, 2020-11-16:
@dan-rodrigues: What diode insertion strategy 2 does is that it inserts fake diodes for each pin in each cell in the design. Then, in the end, it runs the antenna checks, and if a pin has an antenna violation, its fake diode is replaced by a real diode, and then the GDS is generated.

However, for the current version of the skywater-pdk, the libraries don't have fake diodes. So, you won't be able to use this strategy with the skywater-pdk. We were planning on adding a custom cell to the open_pdks setup. Still, it's not very high in our list of priorities given that there is really much gain between strategy 2 or 1 (for all that matters, strategy 1 is much more reliable) because you are inserting the same number of extra cells and compromising the same area. However, strategy 2 is as reliable as the antenna checker could be. On the other hand, strategy 1 is as reliable as spraying diodes at each pin regardless of the antenna checker.

I will add this to the FAQs, and it's up to you to close this issue or leave it open until the custom cells are added somewhere.

dan-rodrigues, 2020-11-17:
Thanks @agorararmard, I tried strategy 1 and it does progress further although I get quite a few overlap errors during `or_diodes.tcl`: `Warning: Overlap check failed (226).`.

This isn't totally related to the original issue so I can review the config.tcl file further or open another issue if this isn't expected behaviour in the `spm` example, if needed

agorararmard, 2020-11-17:
@dan-rodrigues: As I explained above, strategy 1 and 2 insert a diode for each pin, which would at least require double the area (reduced FP_CORE_UTIL and PL_TARGET_DENSITY), and double here is a terrible underestimation. Thus, you need to significantly reduce the FP_CORE_UTIL and PL_TARGET_DENSITY if you want to use diode insertion strategy 1. (Hint: that's why the default is strategy 3 because you don't have to give away all that area and utilization.)

agorararmard, 2020-12-07:
@dan-rodrigues: fake_diodes were added for the hd library on open_pdks and the version was updated in master. I'm closing this issue as it's now resolved. Please, reopen if you find otherwise.

closed 2020-12-07 by agorararmard