The-OpenROAD-Project/OpenLane #91: libs.ref/sky130_fd_io/verilog/sky130_fd_io.v missing

mballance, 2020-11-15:
Hello,
   I'm trying to run some tests for the caravel (eg veriog/dv/caravel/mgmt_soc/uart). These tests seem to expect some verilog model files, such as libs.ref/sky130_fd_io/verilog/sky130_fd_io.v, to exist inside ${PDK_ROOT}/sky130A. These files do not exist anywhere in my ${PDK_ROOT}, and I've tried rebuilding just in case.
    Has something changed here that needs to be updated in caravel? Is something going wrong with my PDK build? Any help would be much appreciated!

Thanks and Best Regards,
Matthew

mballance, 2020-11-15:
BTW, I'm looking at the sky130A_install.log. I think everything mostly looks okay, but I do see a series of messages like the following. Is this a problem?

Error message output from magic:
CIF file read warning: CIF style sky130: units rescaled by factor of 5 / 1
CIF file read warning: Input off lambda grid by 3/5; snapped to grid.
Can't write file named '(UNNAMED)'
Getting GDS file list from /project/fun/openlane/pdk_root/open_pdks/sky130/sky130A/libs.ref/sky130_fd_sc_hdll/gds.
Creating magic generation script to generate magic database files.
No source for abstract views:  Abstract views not made.
Running magic to create magic database files.

Magic 8.3 revision 82 - Compiled on Sat Nov 14 14:38:13 PST 2020.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using NULL graphics device.
Processing system .magicrc file
Sourcing design .magicrc for technology sky130A ...
2 Magic internal units = 1 Lambda
Input style sky130: scaleFactor=2, multiplier=2
Scaled tech values by 2 / 1 to match internal grid scaling
Loading sky130A Device Generator Menu ...
Using technology "sky130A", version 20200508
Writing '(UNNAMED)'
Must specify name for cell (UNNAMED).
Done.

mithro, 2020-11-15:
FYI - @mkkassem @shalan @agorararmard

agorararmard, 2020-11-16:
@mballance: I regenerated the PDK 3 times on two different machines and I always end up with the sky130_fd_io.v being there. However, more and more people are facing the same issue as you on the slack channel. So, I'm not sure what's the issue here, maybe a missing dependency that we're unaware of?
Could you try installing open_pdks through the OpenLANE Makefile with the following commands:
```
make open_pdks
docker run -it -v $RUN_ROOT:/openLANE_flow -v $PDK_ROOT:$PDK_ROOT -e PDK_ROOT=$PDK_ROOT -u $(id -u $USER):$(id -g $USER) openlane:rc4  bash -c "make build-pdk"
```

For both comments, could you open 2 issues here: https://github.com/RTimothyEdwards/open_pdks/issues?

agorararmard, 2020-11-16:
@mballance: I forgot to ask, which versions of open_pdks and skywater-pdk are you using and which branch/tag of openlane?

mballance, 2020-11-17:
Hi @agorararmard,
    Appreciate your quick response! Hmm... Okay, so it looks like this is a strange one. I tried installing open_pdks using the docker image, and experienced the same issue. I was wondering if there was an issue with the 'magic' version I had. I think using the version installed in the docker image rules that out.
    Looking through the console output from the 'make build-pdk' command, I see the following:
```
Writing all magic database files^M
Writing 'sky130_fd_pr__cap_vpp_08p6x07p8_m1m2_lishield'^M
Writing 'sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3m4_shieldl1m5'^M
Writing 'sky130_fd_pr__cap_vpp_11p5x11p7_m1m2_noshield'^M
Writing 'sky130_fd_pr__rf_test_coil1'^M
Writing 'sky130_fd_pr__rf_test_coil2'^M
Writing 'sky130_fd_pr__rf_npn_05v5_W1p00L2p00'^M
Writing 'sky130_fd_pr__rf_test_coil3'^M
Writing 'sky130_fd_pr__cap_vpp_04p4x04p6_m1m2_lishield'^M
Writing 'sky130_fd_pr__pnp_05v5_W3p40L3p40'^M
Writing 'sky130_fd_pr__rf_npn_05v5_W1p00L1p00'^M
Writing '(UNNAMED)'^M
Must specify name for cell (UNNAMED).^M
Done.^M
Error message output from magic:^M
File sky130_fd_pr__rf_npn_05v5_W1p00L1p00.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__rf_npn_05v5_W1p00L2p00.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__pnp_05v5_W3p40L3p40.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__rf_test_coil1.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__rf_test_coil2.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__rf_test_coil3.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__cap_vpp_11p5x11p7_m1m2m3m4_shieldl1m5.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__cap_vpp_11p5x11p7_m1m2_noshield.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__cap_vpp_08p6x07p8_m1m2_lishield.mag couldn't be read^M
No such file or directory^M
File sky130_fd_pr__cap_vpp_04p4x04p6_m1m2_lishield.mag couldn't be read^M
No such file or directory^M
Can't write file named '(UNNAMED)'^M
Migrating LEF files to layout.^M
Warning:  No LEF files found in /project/fun/openlane/pdk_root/open_pdks/sky130/sky130A/libs.ref/sky130_fd_pr/lef^M
```
I did a quick search, and at least the first few of the missing files do exist. Not sure if this provides any clue...

With respect to versions, I'm using:
- OpenLane: rc4 branch
- Skywater commit: 5cd70ed19fee8ea37c4e8dbd5c5c3eaa9886dd23 (from the openlane makefile)
- OpenPDK commit: 48db3e1a428ae16f5d4c86e0b7679656cf8afe3d (from the openlane makefile)

And, yes, I can open an issue on open_pdks.

Thanks again for your help!

Best Regards,
Matthew

agorararmard, 2020-11-17:
@mballance: So, here you're using a version of skywater pdk that doesn't have the IOs, with a version of open_pdks that cannot compile them properly (even if they were there).

Openlane:rc4 is not completely compatible with caravel (and doesn't claim to be so), so it doesn't have the corresponding pdk version, and it lacks some features required for the production. If you want to contribute to the development process, you should be using the branch `develop`; otherwise, since `Caravel` has not yet been produced, we haven't yet released a version of OpenLANE that is capable of reproducing it.

I will leave the decision of whether or not to close the issue up to you.

mballance, 2020-11-17:
Hi @agorararmard,
    Okay, appreciate the response on the state of various elements of the flow. I've checked out and built the 'develop' branch, and this looks much better. Thanks again for your help!

Best Regards,
Matthew

closed 2020-11-17 by mballance