The-OpenROAD-Project/OpenLane #94: opendp seg fault on Caravel DFFRAM module

diecaptain, 2020-11-17:
Hello,

I have an segmentation fault violation during the detailed placement phase while using DFFRAM from caravel under USE_CUSTOM_DFFRAM macro with 256 mem words. The log says that the def output file was successfully written but I have observed that the file is not fully written. The flow then moves on to placement check and ends with a segmentation fault and exit error code 1. The error log and opendp log are attached here.

[opendp.log](https://github.com/efabless/openlane/files/5550435/opendp.log)
[error.log](https://github.com/efabless/openlane/files/5550436/error.log)

The resultant placement def file is attached here for your reference.

[DFFRAM.placement.def.gz](https://github.com/efabless/openlane/files/5550481/DFFRAM.placement.def.gz)

I am using RC4 with relevant open_pdk and skywater tags based on the flow. It is the same for both local docker image and efabless/openlane:rc4 docker image. The config, pdn and pin order files are same as those present in caravel.

I am sure the DFFRAM verilog file in caravel was implemented as is in the macro posted. Verilog wise, the pass mux might be the root of all problems, the way it is implemented.  But I am posting this issue here because the placement def file output is more puzzling to me. The Replace def file in temp directory is written fully. Why is the def file not being completely written and how is the flow deciding that it has been successfully written and proceeding to placement checks? Kindly let me know if you are able to replicate this issue. I am unable to figure out why this behavior is being exhibited.

Thank you.

agorararmard, 2020-11-17:
@diecaptain: Can you share the setup that you used (src + config.tcl) in order for us to try and replicate that error?

P.S: For future reference, rc4 is not completely compatible with caravel (in terms of the pdk version and some minor features).

diecaptain, 2020-11-17:
@agorararmard The source files are here: 

DFFRAM : https://github.com/efabless/caravel/blob/release/verilog/rtl/DFFRAM.v
DFFRAM Building Block : https://github.com/efabless/caravel/blob/release/verilog/rtl/DFFRAMBB.v
Definition File : https://github.com/efabless/caravel/blob/release/verilog/rtl/defines.v

Config file : https://github.com/efabless/caravel/blob/release/openlane/DFFRAM/config.tcl
pdn file : https://github.com/efabless/caravel/blob/release/openlane/DFFRAM/pdn.tcl
pin order file : https://github.com/efabless/caravel/blob/release/openlane/DFFRAM/pin_order.cfg

I was of the opinion that the only missing part is IO cells from the sky130A PDK. Do I use a different commit for the PDK and Open_PDKs to have similar standard cells and lef files apart from IO cells? I moved to RC4 yesterday and so far have come across differences between synthesized netlist and routed netlist resulting in massive LVS net difference and net count mismatches but haven't found time to look at the reason yet. I was happy with RC3 but the non interactive mode was not merging macro lef files provided through EXTRA_LEFS so was running interactively by doing it using add_lefs script, adding obstructions during floorplanning and then placing macros between global placement and IO placement. I saw a recent commit regarding macro placement support and decided to move. Though I am yet to reach that part. DFFRAM was my first test with RC4 after a clean spm run and have come across this issue. 

I can confirm this issue is popping up only because of how pass mux is implemented in the DFFRAM building block. This goes away if it is commented out or replaced with a mux with one input connected to lo and select bit set to lo or hi and by letting yosys optimize it away. I will try and open an issue in caravel regarding the verilog part where its better suited. However it would be nice to know why opendp exhibits this behavior. This is not a huge problem anyway, so I request you to look at this in your spare time.

Thank you.

agorararmard, 2020-11-18:
@diecaptain: I was unable to reproduce the opendp issue locally. I tried master rc4 as well as develop and both worked fine up to routing (I didn't check what happens after).
To answer your question, Caravel is using the latest skywater-pdk and open_pdks. However, Regenerating the blocks in Caravel won't go smoothly for you (it is still a work-in-progress). Also, they are developed using the openlane develop branch which is continuously changing. There should be a new release of OpenLANE as soon as the finalized caravel chip is out.

diecaptain, 2020-11-19:
@agorararmard Thank you for your time. I just found it interesting that this happened.

I am just trying to utilize Caravel as a base now to understand why it is designed the way it is and make an inference. It is interesting. I look forward to finalized caravel chip.

I will close this issue now.

closed 2020-11-19 by diecaptain