The-OpenROAD-Project/OpenLane #95: Parameterised black box fails verilog2def

growly, 2020-11-17:
We have a design that instantiates a submodule. When converting that module to a hard block, we leave the parameters in:

```
mac_cluster #(
  .MAC_CONF_WIDTH(MAC_CONF_WIDTH),  /* this is the error line */
  .MAC_MIN_WIDTH(MAC_MIN_WIDTH),
  .MAC_MULT_WIDTH(MAC_MULT_WIDTH),
  .MAC_ACC_WIDTH(MAC_ACC_WIDTH),
  .MAC_INT_WIDTH(MAC_INT_WIDTH)
) macaroni (
  .clk(clk),
  .rst(rst),
  .en(en),

  .cfg(mac_conf_bus),
  .cset(mac_cset),

  .A0(mac_a0),
  .B0(mac_b0),
  .A1(mac_a1),
  .B1(mac_b1),
  .A2(mac_a2),
  .B2(mac_b2),
  .A3(mac_a3),
  .B3(mac_b3),

  .out0(mac_out0),
  .out1(mac_out1),
  .out2(mac_out2),
  .out3(mac_out3)
);
```

Even though the `mac_cluster` module is marked as `(* blackbox *)`, Yosys generates this instantiation in the output verilog:

```
  mac_cluster #(
    .MAC_ACC_WIDTH(32'sb00000000000000000000000000100000),
    .MAC_CONF_WIDTH(32'sb00000000000000000000000000000100),
    .MAC_INT_WIDTH(32'sb00000000000000000000000000101000),
    .MAC_MIN_WIDTH(32'sb00000000000000000000000000001000),
    .MAC_MULT_WIDTH(32'sb00000000000000000000000000010000)
  ) macaroni ( /* redacted */ );
```

The parameters appear to be a problem for `verilog2def`, which emits this error to its log and then fails to include the module in the output DEF:

```
Error: /openLANE_flow/designs/250_mac_tile/runs/debug/results/synthesis/mac_tile.synthesis.v, line 165106 syntax error, unexpected '.'.
```

However, the flow continues, later failing when the macro cannot be found to place.

```
Placing the following macros:
{'macaroni': ['5000', '5000', 'N']}
Design name: mac_tile
Traceback (most recent call last):
  File "/openLANE_flow/scripts/manual_macro_place.py", line 115, in <module>
    assert not macros, ("Macros not found:", macros)
AssertionError: ('Macros not found:', {'macaroni': ['5000', '5000', 'N']})
```

- It doesn't make sense to parameterise a hardened macro
- It's annoying to have to comment out parameters
- A sensible error should be emited when verilog2def fails like this

agorararmard, 2020-11-17:
@growly: I have to ask, would it be possible to share that design?

ax3ghazy, 2020-11-17:
@growly: I have already addressed this issue in the develop branch; you need to add `-defparam` to the `write_verilog` command (e.g., https://github.com/efabless/openlane/blob/develop/scripts/synth_top.tcl#L66).

growly, 2020-11-17:
Thanks! I pulled the latest develop, and I see the `-defparam` in `synth_top.tcl`:

```
$ tail -n 1 scripts/synth_top.tcl 
write_verilog -noattr -noexpr -nohex -nodec -defparam "$::env(SAVE_NETLIST)"
```

However, I get these errors in what appears to be OpenSTA?
```
[INFO]: Running Static Timing Analysis...
OpenSTA 2.2.0 0d73b5b65a Copyright (c) 2019, Parallax Software, Inc.
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details. 
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
Error: /openLANE_flow/designs/250_mac_tile/runs/debug/results/synthesis/mac_tile.synthesis.v, line 165124 syntax error, unexpected $undefined, expecting ';' or ','.
Error: /openLANE_flow/designs/250_mac_tile/runs/debug/results/synthesis/mac_tile.synthesis.v, line 165125 syntax error, unexpected $undefined, expecting ';' or ','.
Error: /openLANE_flow/designs/250_mac_tile/runs/debug/results/synthesis/mac_tile.synthesis.v, line 165126 syntax error, unexpected $undefined, expecting ';' or ','.
Error: mac_tile is not a verilog module.
create_clock [get_ports $::env(CLOCK_PORT)]  -name $::env(CLOCK_PORT)  -period $::env(CLOCK_PERIOD)
Error: no network has been linked.
Error: no network has been linked.
```

The offending lines of the output verilog are:
```
$ awk 'NR >= 165124 && NR <= 165126' designs/250_mac_tile/runs/debug/results/synthesis/mac_tile.synthesis.v 
  defparam macaroni.MAC_ACC_WIDTH = 32'sb00000000000000000000000000100000;
  defparam macaroni.MAC_CONF_WIDTH = 32'sb00000000000000000000000000000100;
  defparam macaroni.MAC_INT_WIDTH = 32'sb00000000000000000000000000101000;
```

This is the design config: https://github.com/growly/openlane/tree/develop/designs/fpga_250/asic_config/mac_tile
The top module invoking the parameterised modules is: 
 https://github.com/growly/openlane/blob/develop/designs/fpga_250/src/mac_tile.v

diecaptain, 2020-11-21:
@growly Looks like OpenSTA verilog writer is unable to parse defparam syntax in the netlist. Try and see if OpenSTA verilog writer has the capability to do this.

mattvenn, 2020-11-23:
I also have  this issue

agorararmard, 2020-12-07:
@mattvenn, @growly: I implemented a hacky solution for this in the commit referenced above. This was merged to master. I'm closing this issue as it's now resolved. Please, reopen if you find otherwise.

closed 2020-12-07 by agorararmard