The-OpenROAD-Project/OpenLane #97: Map tri-state buffers

growly, 2020-11-19:
Yosys is not scripted to infer and map tri-state buffers. Based on a quick look at the solution implemented to #40, a similar scheme can be used to enable tri-state synthesis.

I modified `scripts/synth.tcl` as follows:

```
diff --git a/scripts/synth.tcl b/scripts/synth.tcl
index 3f9ef60..bdfa203 100755
--- a/scripts/synth.tcl
+++ b/scripts/synth.tcl
@@ -169,6 +169,16 @@ select -clear
 
 hierarchy -check -top $vtop
 
+# Infer tri-state buffers.
+set tbuf_map false
+if { [info exists ::env(TRISTATE_BUFFER_MAP)] } {
+        if { [file exists $::env(TRISTATE_BUFFER_MAP)] } {
+                set tbuf_map true
+                tribuf
+        } else {
+          puts "WARNING: TRISTATE_BUFFER_MAP is defined but could not be found: $::env(TRISTATE_BUFFER_MAP)"
+        }
+}
 
 if { $::env(SYNTH_NO_FLAT) } {
        synth -top $vtop
@@ -182,6 +192,13 @@ opt_clean -purge
 
 tee -o "$::env(yosys_report_file_tag)_pre.stat" stat
 
+# Map tri-state buffers.
+if { $tbuf_map } {
+        puts {mapping tbuf}
+        techmap -map $::env(TRISTATE_BUFFER_MAP)
+        simplemap
+}
+
 # handle technology mapping of latches
 if { [info exists ::env(SYNTH_LATCH_MAP)] && [file exists $::env(SYNTH_LATCH_MAP)] } {
        techmap -map $::env(SYNTH_LATCH_MAP)
```

and then set `::env(TRISTATE_BUFFER_MAP)` to a simple mapping file in my `config.tcl`, using a (for example) Skywater 130 tri-state buffer cell:

```
module \$_TBUF_ (input A, input E, output Y);
  sky130_fd_sc_hd__ebufn _TECHMAP_REPLACE_ (
    .A(A),
    .Z(Z),
    .TE_B(~E));
endmodule
```

Note that this requires a `simplemap` to be run to map the `$not` generated by `~E`.

agorararmard, 2020-11-23:
@growly: Maybe consider PRing this? The OpenLANE part here and the PDK part here: https://github.com/RTimothyEdwards/open_pdks. It will be much faster than waiting for us to do it once we're free. However, we will do it eventually, if you're uncomfortable doing the PR yourself. Thanks!

agorararmard, 2020-11-30:
@growly: This will be merged as soon as the CI tests succeed, #112. the open_pdks update will happen overnight, I believe.

agorararmard, 2020-12-07:
@growly: Merged to master. I'm closing this issue as it's now resolved. Please, reopen if you find otherwise.

closed 2020-12-07 by agorararmard