The-OpenROAD-Project/OpenLane #100: ABC/Yosys does not map muxes

ryanthornton, 2020-11-23:
Similar to #40 and #97, Yosys is not scripted to infer muxes.  Changes similar to the linked issues allow Yosys and ABC to infer and map 2 and 4 muxes.
In `${OPENLANE_ROOT}/scripts/synth.tcl`, I added the following 

```
# handle technology mapping of 4-MUX, and tell Yosys to infer 4-muxes
if { [info exists ::env(SYNTH_MUX4_MAP)] && [file exists $::env(SYNTH_MUX4_MAP)] } {
  muxcover -mux4 
  techmap -map $::env(SYNTH_MUX4_MAP)
  simplemap
}

# handle technology mapping of 2-MUX
if { [info exists ::env(SYNTH_MUX_MAP)] && [file exists $::env(SYNTH_MUX_MAP)] } {
  techmap -map $::env(SYNTH_MUX_MAP)
  simplemap
}
```

Create `${PDK_ROOT}/$::env(PDK)/libs.tech/openlane/$::env(STD_CELL_LIBRARY)/mux_map.v` and `${PDK_ROOT}/$::env(PDK)/libs.tech/openlane/$::env(STD_CELL_LIBRARY)/mux4_map.v` with the following modules: (replacing the technology mapping as desired)
mux_map.v: 
```
module \$_MUX_ (
    output Y,
    input A,
    input B,
    input S
    );
  sky130_fd_sc_hd__mux2_1 _TECHMAP_MUX (
      .X(Y),
      .A0(A),
      .A1(B),
      .S(S)
  );
endmodule
```
mux4_map.v:
```
module \$_MUX4_ (
    output Y,
    input A,
    input B,
    input C,
    input D,
    input S,
    input T
    );
  sky130_fd_sc_hd__mux4_1 _TECHMAP_MUX4 (
      .X(Y),
      .A0(A),
      .A1(B),
      .A2(C),
      .A3(D),
      .S0(S),
      .S1(T)
  );
```
endmodule
In `${PDK_ROOT}/$::env(PDK)/libs.tech/openlane/config.tcl`, add
```
# MUX4 mapping
set ::env(SYNTH_MUX4_MAP) "$::env(PDK_ROOT)/$::env(PDK)/libs.tech/openlane/$::env(STD_CELL_LIBRARY)/mux4_map.v"

# MUX2 mapping
set ::env(SYNTH_MUX_MAP) "$::env(PDK_ROOT)/$::env(PDK)/libs.tech/openlane/$::env(STD_CELL_LIBRARY)/mux_map.v"
```
Finally, the desired muxes must be removed from all appropriate no_synth.cells files.

agorararmard, 2020-11-23:
@ryanthornton: Maybe consider PRing this? The openlane part here and the pdk part here: https://github.com/RTimothyEdwards/open_pdks. It will be much faster than waiting for us to do it once we're free. However, we will do it eventually, if you're uncomfortable doing the PR yourself. Thanks!

shalan, 2020-11-23:
It is worth mentioning that sky130_fd_sc_hd__mux4_* and sky130_fd_sc_hd__mux2_* are too slow. For example, sky130_fd_sc_hd__mux4_1 propagation delay (typical corner) is ~450ps driving the smallest inverter.

ryanthornton, 2020-11-23:
The delay from the sky130_fd_sc_hd__mux* is significantly more than that of the gates Yosys currently replaces _MUX_ primitives with?   There is a vast area reduction by mapping these muxes vs. using sets of ands/ors/etc, so designs favoring density may benefit from the mappings.

closed 2021-02-04 by ax3ghazy