The-OpenROAD-Project/OpenLane #101: RePlAce crash

thesourcerer8, 2020-11-23:
I am trying to build my caravel based design: https://github.com/thesourcerer8/caravel-stdcelllib-stdcells/ and I am getting a crash in the placement:
This is the logfile: http://pdk.libresilicon.com/caravel-build4.html

This is my project directory which hopefully contains all necessary files:
https://pdk.libresilicon.com/user_proj_example.tgz

I have also reported this crash directly to OpenROAD:
https://github.com/The-OpenROAD-Project/OpenROAD/issues/532

I would suggest that we build a bug-reporting tool that can recreate crashes in single steps and put all necessary files and commands together to reproduce the issue.

thesourcerer8, 2020-11-23:
I am thinking that openlane already generates the cmds.log, which puts the commands there, but it is missing the environment variables necessary to reproduce it. What about adding the environment variables into the cmds.log (or a seperate logfile/script) in a way that you can run that script and that it executes the used commands in the same way?

diecaptain, 2020-11-24:
Check the size of your cells and compare it to Bin Size. It feels like your cells are bigger than the bin size resulting in zero whitespace. Nesterov Placer inflates the cell size based on available whitespace and since your whitespace might be zero, the tool exited trying to inflate without success, the utilization is not helping. Your HPWL increased about twice during initial place, so your target density is also not helping. It might help to prune your config based on your design requirement instead of using what is in the caravel config.

thesourcerer8, 2020-11-29:
I solved this crash on my side, I think it was related to either the SIZE attribute in LEF files being missing, or being scaled wrongly (We had accidently written the LEF files in nanometers, but LEF needs to be in microns, so all the numbers were 1000x too big). We will provide test-cases after the tapeout...

agorararmard, 2020-12-07:
@thesourcerer8: Can we close this issue?

donn, 2021-05-04:
Feel free to create a new issue should this arise again.

closed 2021-05-04 by donn