The-OpenROAD-Project/OpenLane #102: OpenSTA fatal error

jean-m-cyr, 2020-11-25:
Running latest openlane and PDKs, trying to build my project with
`./flow.tcl -design sha3_miner`
I get
`[INFO]: Running Static Timing Analysis...
OpenSTA 2.2.0 0d73b5b65a Copyright (c) 2019, Parallax Software, Inc.
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details.
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
Warning: /home/jcyr/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__ff_n40C_1v95.lib, line 31 default_operating_condition ff_n40C_1v95 not found.
Warning: /home/jcyr/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__ss_100C_1v60.lib, line 32 default_operating_condition ss_100C_1v60 not found.
Error: cannot read file /openLANE_flow/openlane/designs/sha3_miner/runs/25-11_06-35/results/synthesis/sha3_miner.synthesis_optimized.v.
`
Looking at /home/jcyr/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__ff_n40C_1v95.lib, we have:
`    default_leakage_power_density : 0.0000000000;
    default_operating_conditions : "ff_n40C_1v95";
    operating_conditions ("ff_n40C_1v95") {
        voltage : 1.9500000000;
        process : 1.0000000000;
        temperature : -40.00000000;
        tree_type : "balanced_tree";
    }
    power_lut_template ("power_inputs_1") {
`
Moving default_operating_conditions like such, and doing the same for /home/jcyr/pdks/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__ss_100C_1v60.lib:

`    default_leakage_power_density : 0.0000000000;
    operating_conditions ("ff_n40C_1v95") {
        voltage : 1.9500000000;
        process : 1.0000000000;
        temperature : -40.00000000;
        tree_type : "balanced_tree";
    }
    default_operating_conditions : "ff_n40C_1v95";
    power_lut_template ("power_inputs_1") {
`
Fixes the two warnings, but the missing /openLANE_flow/openlane/designs/sha3_miner/runs/25-11_06-35/results/synthesis/sha3_miner.synthesis_optimized.v error remains a show stopper. Any ideas why it isn't being created?

agorararmard, 2020-11-25:
@jean-m-cyr: could you share the log of OpenPhySyn (found under `logs/placement`)?

agorararmard, 2020-12-09:
@jean-m-cyr: Any follow-up on that?

agorararmard, 2020-12-21:
@jean-m-cyr: Given the current state of your project, I'll assume this issue has been resolved. Please re-open if you believe otherwise.

closed 2020-12-21 by agorararmard