The-OpenROAD-Project/OpenLane #103: yosys dislikes newlines in liberty files

thesourcerer8, 2020-11-26:
Since the commit
https://github.com/efabless/openlane/commit/6b6a3a20ae0b2e08b92c0cca38d4aa376748ce2a
I got and error from openlane with my liberty file:
1. Executing Liberty frontend.
Imported 428 cell types from liberty file.2. Executing Liberty frontend.
ERROR: Syntax error in liberty file on line 29.
  Unexpected newline.
[ERROR]: during executing: "yosys -c /openLANE_flow/scripts/synth.tcl -l /project/openlane/user_proj_example/runs/user_proj_example/logs/synthesis/yosys.log |& tee >&@stdout"
[ERROR]: Exit code: 1
The whole build log is available here:
https://pdk.libresilicon.com/caravel-build6.html
I am not sure whether the commit actually enabled the liberty file to be parsed at all, or whether it somehow stopped formatting the files properly when merging them.

ax3ghazy, 2020-11-26:
@thesourcerer8: Can you share the `tmp/trimmed.lib` file?

donn, 2021-05-04:
Long inactivity, closed. Feel free to open a new issue if the problem persists.

closed 2021-05-04 by donn