The-OpenROAD-Project/OpenLane #104: bash dependency

thesourcerer8, 2020-11-26:
At least one of the scripts of openlane is relying on features of /bin/bash, but it actually uses #!/bin/sh  which is a symlink to /bin/dash on Ubuntu systems, which makes this function fail and error:
/bin/sh: 1: [[: not found

I would suggest to change all scripts from /bin/sh to /bin/bash to fix this.

smunaut, 2020-11-26:
FYI That's been fixed in a branch already.

agorararmard, 2020-11-26:
@thesourcerer8: I believe this is a duplicate issue for #82. Am I correct?

smunaut, 2020-11-26:
https://github.com/efabless/caravel/commit/4666810f4506809d4f3fb1aaaa57d96edfaef385

thesourcerer8, 2020-11-27:
No, it's the same kind of problem in several places I would say. I have provided a fix in #105

agorararmard, 2020-11-30:
@thesourcerer8: This should be fixed since #109 was merged. However, I'll keep this issue open until the fix reaches `master`.

agorararmard, 2020-12-07:
@thesourcerer8: Merged to master. I'm closing this issue as it's now resolved. Please, reopen if you find otherwise.

closed 2020-12-07 by agorararmard