The-OpenROAD-Project/OpenLane #113: SELECT ROUTES

soorajkvl, 2020-12-01:
The OpenLANE helps to include standard cells but not possible to include different LEF routing.

Example: Single die with PLL,USB,PICORV32A and routing between PLL and PICORV32A or USB and PICORV32A

Please help to know how is possible to route selective cells.

agorararmard, 2020-12-02:
@soorajkvl: Can you elaborate more on your question? Do you mean you have 3 macros and you want to route signals between them? I'm not familiar with the term "LEF routing", so can you explain it to me or point me to some source?

soorajkvl, 2020-12-03:
@agorararmard : Your guess is right. I have three macros (or Library Extension Formate .lef layout files) and need to do placement and route between them in OpenLANE.

How is that possible?

agorararmard, 2020-12-04:
@soorajkvl: Check this directory https://github.com/efabless/openlane/tree/master/designs/manual_macro_placement_test

agorararmard, 2020-12-09:
@soorajkvl: Did this answer your question?

soorajkvl, 2020-12-09:
Yes, Thank you so much.

closed 2020-12-09 by soorajkvl