The-OpenROAD-Project/OpenLane #114: padframe genaration error

rsnkhatri3, 2020-12-02:
I am keeping the padframe_folder in designs/my_design_folder, but below shown error occurs
[image]
[image]

agorararmard, 2020-12-02:
@rsnkhatri3: We don't officially fully support automated chip integration. However, you can check this directory https://github.com/efabless/caravel/tree/develop-pruned/openlane/chip_io as an example of how to create a padframe. (Expect to do some manual edits in the layout when you go to the chip integration stage).

rsnkhatri3, 2020-12-03:
While using the interactive script from above link, it asks for config.json file what to include in it.

closed 2020-12-09 by rsnkhatri3