The-OpenROAD-Project/OpenLane #115: or_route.tcl segmentation violation on develop branch

SweeperAA, 2020-12-02:
During execution of ./flow.tcl -design <design_name>, the or_route.tcl script faults with the following output:

Update congestion history type 2
[INFO] iteration 42, enlarge 211, costheight 288, threshold 0 via cost 0 
[INFO] log_coef 0.514090, healingTrigger 19 cost_step 2 L 1 cost_type 1 updatetype 2
[Overflow Report] total Usage   : 371661
[Overflow Report] Max H Overflow: 17
[Overflow Report] Max V Overflow: 19
[Overflow Report] Max Overflow  : 19
[Overflow Report] Num Overflow e: 4994
[Overflow Report] H   Overflow  : 13055
[Overflow Report] V   Overflow  : 12771
[Overflow Report] Final Overflow: 25826

Update congestion history type 2
[INF[ERROR]: during executing: "openroad -exit /openLANE_flow/scripts/openroad/or_route.tcl |& tee >&@stdout /openLANE_flow/designs/decred_block/runs/02-12_02-51/logs/routing/fastroute.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: kill signal

[ERROR]: Please check openroad  log file
[ERROR]: Dumping to /openLANE_flow/designs/decred_block/runs/02-12_02-51/error.log

agorararmard, 2020-12-02:
@SweeperAA: I will need more details on that one. However, an initial guess is that your design has too many cells or is too congested. Either way, make sure to have `CELL_PAD` set to 4, and reduce `FP_CORE_UTIL` and `PL_TARGET_DENSITY` for this design. Keeping `PL_TARGET_DENSITY` = (`FP_CORE_UTIL`+ (1~5))/100

Also check: https://github.com/efabless/openlane/wiki#why-am-i-seeing-placement-failed-errors-in-detailed-placement-or-cant-handle-congested-design-during-routing

SweeperAA, 2020-12-02:
@agorararmard The script erroring out like that doesn't give any indication of what is wrong. It's not outputting any warnings or errors about placement or congestion. Depending how I run the flow (either through caravel's `make <design>` or through openlane's docker via `./flow.tcl -design <design>`), I either get `child killed: kill signal` or `child killed: segmentation violation`. At a minimum, I think the tools should handle fault conditions more gracefully than that and give an indication of what is wrong. Even if it is a type of "congestion error". 

This design is about 50k cells and the following are the settings I have in the config.tcl:

`set ::env(FP_SIZING) absolute
set ::env(DIE_AREA) "0 0 2920 3520"
set ::env(DESIGN_IS_CORE) 0
set ::env(FP_PDN_CORE_RING) 0
set ::env(GLB_RT_MAXLAYER) 5
set ::env(FP_CORE_UTIL) "6"
set ::env(PL_TARGET_DENSITY) 0.06
set ::env(SYNTH_STRATEGY) "2"
set ::env(SYNTH_MAX_FANOUT) "6"
set ::env(FP_ASPECT_RATIO) "1"
set ::env(CELL_PAD) "2"
set ::env(GLB_RT_ADJUSTMENT) "0.0"
set ::env(DIODE_INSERTION_STRATEGY) "3"
`

With the `DIE_AREA` defined as above, the design takes up 5.3% of the area so the density can't be reduced much more. Is it really a density or congestion issue with only using 5.3%? Is there any information in the output I provided that indicates magnitude of complexity of the design?

If I increase `CELL_PAD` to 4, I get `Warning: Overlap check failed (54)` and `Warning: Site check failed (1).` from the `or_opendp.tcl` script before the `or_route.tcl` script runs.

So to get past those errors, I'm having to set `CELL_PAD` to 2 (setting `FP_CORE_UTIL` to 2 results in no change either way).

The project files and scripts are all at https://github.com/SweeperAA/skywater130_decred_miner

If there are any log or other input files I can provide you to help debug this issue, let me know. Or if there are other values I should use in my config.tcl script, let me know.

SweeperAA, 2020-12-03:
@agorararmard I varied `SYNTH_STRATEGY` and it got much further in the congestion process and still failed. However, in that case it did give a reasonable error of `Design congestion too high`.

I'll continue to vary other parameters, but `SYNTH_STRATEGY` set to 0 seemed to get me the closest to passing congestion.

agorararmard, 2020-12-09:
@SweeperAA: From slack activity, I think you moved past this issue. However, I should mention that with `set ::env(FP_SIZING) absolute` you can only have `set ::env(DIE_AREA) "0 0 2920 3520"` and `FP_CORE_UTIL` is ignored. On the other hand, if you have `set ::env(FP_SIZING) absolute` you can only have `set ::env(FP_CORE_UTIL) "6"` and `DIE_AREA` will be ignored.

Changing the `SYNTH_STRATEGY` is one way to go around it. You can also try changing `PL_TARGET_DENSITY`.

This might also prove useful: https://github.com/efabless/openlane/wiki#what-to-do-if-the-flow-failed-in-any-of-the-steps-from-floorplan-up-to-routing

SweeperAA, 2020-12-09:
@agorararmard I think what you meant to say is that if `FP_SIZING` is defined as `absolute`, then `FP_CORE_UTIL` is ignored and if `FP_SIZING` is defined as `relative`, then `DIE_AREA` is ignored.

While I have moved past this issue by adjusting parameters (via trial and error), user applications need to check boundary conditions so they can fail gracefully with error codes indicating the problem. Not just die with `child killed: kill signal`.  Without an error code explaining what the problem is, the FAQ you referenced is not helpful in trying to adjust the design.

somayyahugar, 2021-05-19:
[image]
Hi,
sir i'm getting this error while running the openlane for the command run_routing for the simple adder design can anyone help me to resolve

closed 2021-06-16 by donn