The-OpenROAD-Project/OpenLane #119: magic exits during LEF generation with kill signal

SweeperAA, 2020-12-04:
After GDS write completed and during LEF generation, magic exits with a `kill signal`. This occurred in the openlane docker with magic version 8.3.87 on the openlane develop branch (checked out Nov 30). I've provided context of the log below and a link to the routed DEF and `merged.lef` files (the routed DEF was over 400MB so I had to break up into 10MB compressed archive chunks using 7zip).

```
Writing cell sky130_fd_sc_hd__buf_12
[INFO]: GDS Write Complete
[INFO]: Writing abstract LEF
Generating LEF output /openLANE_flow/designs/decred_nonblock/runs/8mFP50DENS50STRAT1_20ns/results/magic/decred_nonblock.lef for cell decred_nonblock:
Diagnostic:  Write LEF header for cell decred_nonblock
Diagnostic:  Writing LEF output for cell decred_nonblock
[ERROR]: during executing: "magic -noconsole -dnull -rcfile /home/dracex/PDK_ROOT/sky130A/libs.tech/magic/sky130A.magicrc /openLANE_flow/scripts/magic.tcl </dev/null |& tee >&@stdout /openLANE_flow/designs/decred_nonblock/runs/8mFP50DENS50STRAT1_20ns/logs/magic/magic.log"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child killed: kill signal
[ERROR]: Please check magic  log file
[ERROR]: Dumping to /openLANE_flow/designs/decred_nonblock/runs/8mFP50DENS50STRAT1_20ns/error.log
    while executing
"try_catch magic  -noconsole  -dnull  -rcfile $::env(MAGIC_MAGICRC)  $::env(SCRIPTS_DIR)/magic.tcl  </dev/null  |& tee $::env(TERMINAL_OUTPUT) $::env(m..."
    (procedure "run_magic" line 14)
    invoked from within
"run_magic"
    (procedure "run_non_interactive_mode" line 27)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 168)
```

Link to files: https://github.com/SweeperAA/skywater130_decred_miner_old/tree/main/DEBUG

agorararmard, 2020-12-09:
@SweeperAA: I believe @ax3ghazy addressed this issue with you elsewhere. Does it still stand?

SweeperAA, 2020-12-09:
Both the design and magic tool have been updated so I don't know if it was ever fixed, but I'm not running into this problem anymore.

closed 2020-12-09 by SweeperAA