The-OpenROAD-Project/OpenLane #123: ERROR: Can't open ABC output file `/tmp/yosys-abc-usQczk/output.blif'.

SweeperAA, 2020-12-06:
Using openlane mpw-one-a branch and in the early stages of the flow for my design, I get the following error:

```
22.1.1. Executing ABC.
ERROR: Can't open ABC output file `/tmp/yosys-abc-usQczk/output.blif'.
[ERROR]: during executing: "yosys -c /openLANE_flow/scripts/synth.tcl -l /project/openlane/decred_top/runs/decred_top/logs/synthesis/yosys.log |& tee >&@stdout"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
child process exited abnormally

[ERROR]: Please check yosys  log file
[ERROR]: Dumping to /project/openlane/decred_top/runs/decred_top/error.log

    while executing
"try_catch [get_yosys_bin]  -c $::env(SYNTH_SCRIPT)  -l $::env(yosys_log_file_tag).log  |& tee $::env(TERMINAL_OUTPUT)"
    (procedure "run_yosys" line 22)
    invoked from within
"run_yosys"
    (procedure "run_synthesis" line 4)
    invoked from within
"run_synthesis"
    (procedure "run_non_interactive_mode" line 11)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "/openLANE_flow/flow.tcl" line 162)
Makefile:25: recipe for target 'decred_top' failed
make: *** [decred_top] Error 1
```

This error occurs on the full up version of my design that as about 282K cells (as reported just before the "Executing ABC pass (technology mapping using ABC)" step). If I reduce my design to 107K cells, it still occurs. Only if I reduce the design down to 72K cells, I don't get this error. I don't believe this is a RAM limitation as the memory usage was very low during this part of the process (and I have 24G of RAM).

Repo is at: https://github.com/SweeperAA/skywater130_decred_miner

SweeperAA, 2020-12-07:
Another data point is that my design uses `SYNTH_STRATEGY` 1. If I change to 2 or 3, I do not get this error. A value of 0 still produces the error.

Also, the github repo link changed per mpw-one guidelines of caravel_ prefix. Github will redirect, but for reference, it is now https://github.com/SweeperAA/caravel_skywater130_decred_miner

SweeperAA, 2020-12-08:
In the mean time, I've updated my repo for a smaller design to get around this issue. If you need to recreate it with the repo I listed, let me know and I'll provide you the changes to do so.

agorararmard, 2020-12-09:
@SweeperAA: Usually, when this error happens: `Can't open ABC output file `/tmp/yosys-abc-usQczk/output.blif'` it means one of two things:
- Your root directory is full of yosys and magic crash databases and needs to be cleaned up (or you can close and re-start the docker).
- ABC failed to map your design, and the solution is indeed to try a different SYNTH_STRATEGY (that's what they are there for.) in order for your design to get mapped.

Since you already tried the other synthesis strategies and they worked. I believe your problem is solved.

SweeperAA, 2020-12-09:
@agorararmard I am working around this issue by changing the `SYNTH_STRATEGY`. I would not say that the issue in the tool is addressed by either handling the problematic `SYNTH_STRATEGY` or providing a useable error code.

closed 2021-05-04 by donn