The-OpenROAD-Project/OpenLane #124: Segfault while "Generating PDN..."

dan-rodrigues, 2020-12-08:
Gist with error log:
https://gist.github.com/dan-rodrigues/6593792f87b3dc31780c5a52a1164d74

Commit in repo to reproduce this error:
https://github.com/dan-rodrigues/caravel_vdp_lite/commit/ea3a92393985a52c3b24194fb554234218851c61

Steps to reproduce in the linked repo:

```
cd openlane
make vdp_lite_user_proj
```

Commits used to trigger the error:

* openlane (including built docker image): `82e87c24d495882a39dd4046bb133e95ff8334ba` (`mpw-one-a`)
* open_pdks: `b184e85de7629b8c87087a46b79eb45e7f7cd383` (`mpw-one-a`)

Commits that produce a clean user project and wrapper:

* openlane: `817314b`
* open_pdks: `8688323`

agorararmard, 2020-12-08:
@dan-rodrigues: I believe this might be caused by the fact that you have a `gen_pdn` in your interactive script here https://github.com/dan-rodrigues/caravel_vdp_lite/blob/ea3a92393985a52c3b24194fb554234218851c61/openlane/vdp_lite_user_proj/interactive.tcl#L50
However, the `gen_pdn` step has been moved inside `run_floorplan` in the current openlane master (mpw-one-a). https://github.com/efabless/openlane/blob/82e87c24d495882a39dd4046bb133e95ff8334ba/scripts/tcl_commands/floorplan.tcl#L255

So the solution is to simply remove `gen_pdn` from your script.

P.S: While it's desirable to reproduce your blocks with each significant update of the toolchain, it's not a must. So, you don't have to reharden your blocks. The only two steps that you need to do with each significant update is reharden `user_project_wrapper` and rerun `make ship`/`make`.

dan-rodrigues, 2020-12-09:
Thanks, I've removed the `gen_pdn` step and that resolved the segfault.

closed 2020-12-09 by dan-rodrigues