The-OpenROAD-Project/OpenLane #126: make test failed on rc5

sajjadahmed677, 2020-12-11:
i am trying to setup the openlane in my laptop but facing an issue while executing make test command. here is the error.
couldn't read file "/home/sky130A/libs.tech/openlane/config.tcl": no such file or directory
    while executing
"source $pdk_config"
    (procedure "prep" line 126)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 9)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	puts_info "Running interactively"
	if { [info exists arg_values(-file)..."
    (file "./flow.tcl" line 162)
make: *** [Makefile:113: test] Error 1

agorararmard, 2020-12-11:
@sajjadahmed677: Did you install the pdk in `/home`?

sajjadahmed677, 2020-12-12:
yes, but the error is in make file

agorararmard, 2020-12-16:
@sajjadahmed677: So to clarify, the installation of the PDK was successful and `/home/sky130A/libs.tech/openlane/config.tcl` does exist?

Haseeb-Baloch, 2020-12-18:
> i am trying to setup the openlane in my laptop but facing an issue while executing make test command. here is the error.
> couldn't read file "/home/sky130A/libs.tech/openlane/config.tcl": no such [...]

The error isn't in the makefile the problem is that the Sky130A folder isn't present in the PDK_ROOT you have provided.
Either set the PDK_ROOT to the Sky130A folder present in $PDK_ROOT/sky130/
Or, copy the Sky130A folder to $PDK_ROOT/

agorararmard, 2020-12-21:
@sajjadahmed677: Does this issue still stand?

agorararmard, 2021-01-04:
@sajjadahmed677: I will assume the issue is resolved, and I'll close this. Please re-open if otherwise.

closed 2021-01-04 by agorararmard