The-OpenROAD-Project/OpenLane #132: Pull access denied for openlane

ammar89, 2020-12-19:
While trying to setup openlane, i got the following error message:

```
Unable to find image 'openlane:rc6' locally
docker: Error response from daemon: pull access denied for openlane, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
```

My machine is running Ubuntu 18.04.

I change the `IMAGE_NAME ?= openlane:rc6` (line 21) in the make file to `IMAGE_NAME ?= efabless/openlane`, which solved the issue apparently.

```
Magic DRC Summary:
Source: /openLANE_flow/designs/spm/runs/openlane_test/logs/magic/magic.drc
Total Magic DRC violations is 0
----------------------------------------

LVS Summary:
Source: /openLANE_flow/designs/spm/runs/openlane_test/results/lvs/spm.lvs_parsed.log
LVS reports no net, device, pin, or property mismatches.
Total errors = 0
----------------------------------------

Antenna Summary:
Source: /openLANE_flow/designs/spm/runs/openlane_test/reports/routing/antenna.rpt
Number of pins violated: 0
Number of nets violated: 0
[SUCCESS]: Flow Completed Without Fatal Errors.
Basic test passed
```
I only got two warning messages,

```
[WARNING]: Current core area is too small for a power grid
[WARNING]: Minimizing the power grid!!!!
```

agorararmard, 2020-12-21:
@ammar89: It's not very clear what the issue is here. You used dockerhub to pull the openlane docker instead of building it locally (as per the instructions), then you had to export `IMAGE_NAME` to the new name which is `efabless/openlane:rc6` instead of the locally built image name `openlane:rc6` (as the instructions in the README say). By doing so as per the instrcutions, you were able to run the docker image successfully and get a clean run out of openlane. So, where is the issue or where is the question?

As for the 2 warnings, they are just to tell you that the flow will auto adjust the power grid to allow it to fit inside the design's small core area. Nothing to worry about.

README instructions on dockerhub usage:
https://github.com/efabless/openlane#pulling-an-auto-built-docker-image-from-dockerhub
https://github.com/efabless/openlane#running-the-pulled-auto-built-docker-image

robtaylor, 2020-12-21:
I also can't get `make test` to work:

(base) robtaylor@Roberts-MacBook openlane % make test
cd /Users/robtaylor/Code/Chipflow/openlane && \
		docker run -it -v /Users/robtaylor/Code/Chipflow/openlane:/openLANE_flow -v /Users/robtaylor/Code/Chipflow/:/Users/robtaylor/Code/Chipflow/ -e PDK_ROOT=/Users/robtaylor/Code/Chipflow/ -u 501:20 openlane:rc7 sh -c "./flow.tcl -design spm -tag openlane_test -disable_output -overwrite"
Unable to find image 'openlane:rc7' locally
docker: Error response from daemon: pull access denied for openlane, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.

I've followed the README to the letter.

agorararmard, 2020-12-21:
@robtaylor: Your case is different. When you rebased to develop to submit PR #133 you switched to openlane:rc7 (current develop), while before rebasing you were in master (openlane:rc6). So, you'll have to rerun `make` after rebasing. (or switch back to master in a different branch of your fork). Did I get that right?

ammar89, 2020-12-27:
Thanks agorararmard, it was a mistake from my end, I built openlane locally and it is working fine. I apologize for the inconvenience.

closed 2020-12-27 by ammar89