The-OpenROAD-Project/OpenLane #146: No support for OSX Aarch64

robtaylor, 2021-01-02:
Being the recipient of a new M1 based Mac this Christmas, I'm now hacking up support for AArch64.

Two questions:
1)  is there a reason that we don't use docker compose for building the openlane:rc7 image?
2) Would a multi-arch buildx based solution be acceptable?

robtaylor, 2021-01-02:
Another Q - is there a reason we aren't using multistage builds?

robtaylor, 2021-01-02:
another Q - is centos 7 support important, or could centos 8 be a base?

ax3ghazy, 2021-01-04:
Hi @robtaylor,

> One question: Would a multi-arch BuildKit based solution be acceptable?

Don't see why not; sure:)

> Another Q - is there a reason we aren't using multistage builds?

Nope, just never got around to optimizing the Dockerfiles, but yes, ideally, for example, they should all start from a base image that includes all the common prerequisites. What other specific advantages of multi-stage builds could you spot here?

> another Q - is centos 7 support important, or could centos 8 be a base?

Anything that is not too old (e.g., centos6 by now) can be a base, so centos8 should work fine as a base.

antonblanchard, 2021-01-06:
I'm running it on ppc64le Linux with a hacked up set of Docker files:

https://github.com/antonblanchard/openlane/tree/docker-cross-platform

Would be nice to clean the Docker files up properly. Note I couldn't get CVC to build because pyinstaller is doing very strange things (producing an x86 binary on ppc64le. Odd).

antonblanchard, 2021-01-07:
I agree it would be nice to rebase to centos8.

robtaylor, 2021-01-08:
I’ve got a patch set neatly together. Just wrestling buildkit caching...

robtaylor, 2021-01-08:
I'll also add in ppc64le, see if it works..

WIP here https://github.com/robtaylor/openlane/tree/feature/multiarch

robtaylor, 2021-01-08:
@antonblanchard I see we found many of the same issues, my apologies for not sharing earlier!

robtaylor, 2021-01-10:
Ok. So very nearly there. Completely stumped by this one though, building OpenPhySyn:

#20 5.398   Could NOT find SWIG (missing: 3.0) (found version "3.0.12") 
#20 5.398 Call Stack (most recent call first):
#20 5.398   /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:458 (_FPHSA_FAILURE_MESSAGE)
#20 5.398   /usr/share/cmake/Modules/FindSWIG.cmake:105 (find_package_handle_standard_args)
#20 5.398   external/OpenSTA/CMakeLists.txt:293 (find_package)

its quite bizzare.. Same happens for OpenDB, but note that OpenSTA and OpenDB both compile fine standalone. Any ideas anyone?

WIP here https://github.com/robtaylor/openlane/tree/feature/multiarch

robtaylor, 2021-01-12:
turns out their cmakelists.txt are badly formmatted and failing on newer cmakes. patching..

this is definitely more epic than originally planned!

ax3ghazy, 2021-01-13:
@robtaylor: The error message looked familiar. It was probably this: https://github.com/The-OpenROAD-Project/OpenROAD/issues/499#issuecomment-692824474, wasn't it?

robtaylor, 2021-01-13:
> familiar. It was probably this: The-OpenROAD-Project/OpenROAD#499
> (comment)

Yep, thank you!  Issue is that openphysyn has too old OpenDB and OpenSTA.

Currently testing with moving those sub modules to current head.

donn, 2021-12-05:
December 2021 Status Update

* There's a common centos7 base image for all tools now, but sadly, by the looks of it, aarch64 doesn't appear to be supported in centos7: https://github.com/The-OpenROAD-Project/OpenLane/issues/146
* Given that the CentOS project is dead, it might be prudent to just rebase to Debian.

If you're using an aarch64 Mac you may have to build the tools and images manually, see the following blog post for a starting point: https://blog.jaimyn.dev/how-to-build-multi-architecture-docker-images-on-an-m1-mac/

robtaylor, 2021-12-05:
See my branch above in this thread for centos8 & aarch64 work. It's probably bitrotted a little, but a lot of work was done and you should have plenty to go on.

donn, 2021-12-05:
Yeah, that'll definitely be my starting point once I get my hands on an aarch64 device. Hopefully Rocky Linux is a decent drop-in substitute for centos8.

robtaylor, 2021-12-06:
Worth spinning up an AWS graviton instance!

donn, 2021-12-24:
If I may, what was your solution for Klayout? Building it from source is a certifiable PITA and they don't appear to be doing aarch64 Klayout binaries.

robtaylor, 2021-12-25:
From branch feature/multiarch in https://github.com/robtaylor/openlane

It was a while ago, so this may have broken since then…

‘’’
##################### klayout #######################
FROM base AS klayout

# install klayout
RUN yum install -y qt5-devel ccache
RUN git config --global http.postBuffer 524288000
RUN git clone https://github.com/KLayout/klayout.git && \
    cd klayout && \
    git checkout v0.26.9-2

WORKDIR klayout
RUN mkdir -p /build/lib /build/bin
RUN QMAKE_CCACHE=1 ./build.sh -rpath /usr/local/lib/ -bin /build/bin
-option "-j$(nproc)"
RUN mv /build/bin/lib*.so* /build/lib

RUN /stamp.sh klayout
‘’’

donn, 2022-02-14:
Klayout's been built as an OpenLane tool proper. That was the last obstacle- now I need to just boot up the Graviton instance and test...

EngRaff92, 2022-03-23:
I’m planning to install open over this week, is there any update on the apple silicon support? Thanks for the hard work on getting this up and running

donn, 2022-03-28:
@EngRaff92 Not yet, I've been extremely busy the last while. I'll get to it soon, I promise.

EngRaff92, 2022-03-28:
No worries at all, was just asking since the entire thread was not updated further thanks for confirming regards

cbalint13, 2022-04-08:
@robtaylor 
@ax3ghazy 

FYI

* Complete centos8, fc35, fc36, fc37 x{aarch64, ppc64le, x86_64}  native package support: https://gist.github.com/cbalint13/1de29fbe77e8dedddd3861b5f49a1835

* Mentioned repository are weekly automated (including CI regressions), holds **all tooling needed** for openlane, and beyond.

* In case of **openlane** post-build regressions are done via  ```openlane-flow.tcl -design spm``` see [in logs here](https://download.copr.fedorainfracloud.org/results/rezso/VLSI/epel-8-aarch64/03938276-openlane/builder-live.log.gz)

Cc: @maliberty .

Thank you,
~Cristian.

cbalint13, 2022-04-08:
@robtaylor 
@ax3ghazy 
@donn 

FYI

* Complete centos8, fc35, fc36, fc37 x{aarch64, ppc64le, x86_64}  native package support: https://gist.github.com/cbalint13/1de29fbe77e8dedddd3861b5f49a1835

* The [VLSI repository](https://copr.fedorainfracloud.org/coprs/rezso/VLSI/) is weekly automated (+CI regressions), holds **all tooling needed** for openlane, and beyond.

* In case of **openlane** post-build regressions are done via  ```openlane-flow.tcl -design spm``` see [sample log here](https://download.copr.fedorainfracloud.org/results/rezso/VLSI/epel-8-aarch64/03938276-openlane/builder-live.log.gz)

* Made an announcement on social media (for record): https://twitter.com/cbalint13/status/1509831008627355649

Cc: @maliberty , @RTimothyEdwards
Cc: @antonblanchard (ppc64le)

Not perfect but is a start towards some native releasing.

Thank you,
~Cristian.

donn, 2022-05-21:
It's done. We're still on CentOS 7 though.

closed 2022-05-21 by donn

cbalint13, 2022-06-01:
> Hello just to clarify is this the link we should be referring to if we wanna install it on Apple Silicon? https://gist.github.com/cbalint13/1de29fbe77e8dedddd3861b5f49a1835

@EngRaff92,

* If you run Fedora>=35 or CentOS>=8 (aarch64) on that Apple Sillicon than **YES** that install procedure will work too.

> 
> I'm not quite getting is noarch X86 refers to the arch selected or if jus an example.

That is .rpm (CentOS/Fedora) [rpm package](https://en.wikipedia.org/wiki/RPM_Package_Manager) specific things:
* 'noarch'  is a .rpm package having only data (no executable at all)
* 'x86_64' is a .rpm package targeting x86_64 architecture.

EngRaff92, 2022-06-01:
Hey thanks for getting back nope I’m not running none of the os you mentioned just macOS. So according to what you wrote it will not work properly on this MacOS, will it ? Regards

EngRaff92, 2022-06-01:
Cool thanks for confirming

EngRaff92, 2022-06-01:
I followed the instructions in the README and I installed properly OpenLane. Once I ran the test (make tests) it fails with the following error:

Docker info was not valid JSON.
Traceback (most recent call last):
  File "/Users/raffaelemac/OpenLane/./env.py", line 257, in <module>
    main()
  File "/Users/raffaelemac/OpenLane/./env.py", line 253, in main
    commands[args[0]]()
  File "/Users/raffaelemac/OpenLane/./env.py", line 59, in docker_config
    if cinfo.engine == "docker":
AttributeError: 'NoneType' object has no attribute 'engine'
cd /Users/raffaelemac/OpenLane && \
		docker run --rm -v /Users/raffaelemac/OpenLane:/openlane -v /Users/raffaelemac/OpenLane/designs:/openlane/install -v /Users/raffaelemac/OpenLane/pdks:/Users/raffaelemac/OpenLane/pdks -e PDK_ROOT=/Users/raffaelemac/OpenLane/pdks -e STD_CELL_LIBRARY=sky130_fd_sc_hd  efabless/openlane:e76700234cce061d5d37021214f6453e0e1dcb0e-arm64v8 sh -c "./flow.tcl -design spm -tag openlane_test -overwrite"
Basic test failed

** ANY IDEA **
Regards
Raffaele

PR #1075 (2022-05-21): Support Multi-Arch Docker Builds
+ Add support for multiple-architecture building
    + Support for amd64, ppc64le and arm64v8 added
    + amd64 and arm64v8 will be automatically built by the CI

~ Major CI Overhauls
   ~ Added composite actions for the Docker build and setting environment variables
   ~ pdk build is now its own job, and amd64/aarch64 have a job each
   ~ matrix -> design_matrix, designs only tested on amd64 for performance reasons

~ run/build base docker images now pushed to Docker Hub based on a hash of dependency lists (to work with the buildx `docker-container` driver)

- [... cut]