The-OpenROAD-Project/OpenLane #153: GLB_RT_OBS prevents filler cells from being placed

antonblanchard, 2021-01-07:
I narrowed down the issue I'm seeing with https://github.com/efabless/openlane/pull/143 when building some of my macros. These macros create an obstruction across all of `met5` (otherwise I end up with tracks on `met5` even with `GLB_RT_MAXLAYER` set):

```
set ::env(GLB_RT_MAXLAYER) 5

# Tracks are ending up on met5 even with GLB_RT_MAXLAYER set
set ::env(GLB_RT_OBS) "met5 $::env(DIE_AREA)"
```

It looks like filler cells aren't getting placed at all. Here is what I'd expect it to look like:
[image]

And here is what is I'm seeing:
[image]

antonblanchard, 2021-01-07:
```
[INFO DPL-0001] Placed 0 filler instances.
```

So no filler cells were placed at all. The obstruction is only on met5:
```
MACRO obs_core_obs_0
   CLASS BLOCK ;
   FOREIGN obs_core_obs_0 ;
   ORIGIN 0.000 0.000 ;
   SIZE 670.0 BY 670.0 ;
   OBS
        LAYER met5 ;
             RECT 0.000 0.000 670.0 670.0 ;
   END
 END obs_core_obs_0
```

And the filler cell has nothing on met5 (as you'd expect), so why can't they be placed?

agorararmard, 2021-01-08:
Hi @antonblanchard, thanks for pointing that out. This commit should fix it 41776e8a0b7ce04bb5158ecc745a9b62fcfa746a. Can you try it out and verify that it works on your case?

Note that: 
```
[WARNING]: Specifying a routing obstruction is now done using the coordinates
[WARNING]: of its bounding box instead of the now deprecated (x, y, size_x, size_y).
```
So, now:
```
| `GLB_RT_OBS` | Specifies custom obstruction to be added prior to global routing. Comma separated list of layer and coordinates: `layer llx lly urx ury`.<br> (Example: `li1 0 100 1000 300, met5 0 0 1000 500`)  <br> (Default: unset) |
```

antonblanchard, 2021-01-08:
@agorararmard Thanks! This fixes my issue, and I did trip over the size vs coordinates issue so that makes sense to me too.

antonblanchard, 2021-01-11:
Closing this since it has been verified to work.

closed 2021-01-11 by antonblanchard