The-OpenROAD-Project/OpenLane #156: DRC Viewer

thesourcerer8, 2021-01-13:
The idea is to easily visualize the DRC results after a “make” build.
Instead of running 
make user_proj_example
magic -T ../../$PDK/long/path/sky130A.tech user_proj_example/runs/user_proj_example/results/magic/user_proj_example.drc.mag 
which is a bit long to type, but a make target like “make showdrc” would be the preferred solution:
make user_proj_example
make showdrc

I have implemented such a viewer like this with the command ./showdrc.sh for my caravel project:
https://github.com/thesourcerer8/caravel_stdcelllib_stdcells/commit/46d921915a7acf8cb173c3eb0a414a017a9221eb#diff-46cb03e3da088b910794e550640492118d3b56e74697e5629fcc5342a6d216fa

One possibility is to use magic supplied by the user outside the docker envioronment, the other possibility would be to use the magic installed in the docker container with X-forwarding or something similar.

agorararmard, 2021-01-13:
@thesourcerer8: With the current develop branch, a TCL script is already generated to load the violations into an already open magic viewer with the GDS read. One possibility is to create a bash/TCL script that automatically loads the GDS into magic, with the .magicrc used to create the GDS and then load it into magic. In which case, the user has to have magic available. I'll be doing that in the next couple of days. 

Otherwise, the user can use the already create Klayout XML database to view the violations in his/her local Klayout.

However, this issue should remain open until these changes reach master with rc7.

thesourcerer8, 2021-01-13:
I just tried to use the magic inside the docker container, you just need to add 2 parameters to the docker run command and it works to use the magic from the container:         -e DISPLAY=$(DISPLAY)     -v /tmp/.X11-unix:/tmp/.X11-unix

agorararmard, 2021-01-14:
@thesourcerer8: That would be an optional user preference.

donn, 2022-01-03:
I think I can safely say this reached master at some point.

closed 2022-01-03 by donn