The-OpenROAD-Project/OpenLane #158: Integrate build report tool

thesourcerer8, 2021-01-13:
Please adapt and integrate my build report tool as a standalone tool at the end of the flow (or in a way that it always runs afterwards, even if the flow crashes before)
https://github.com/thesourcerer8/caravel_stdcelllib_stdcells/blob/release/openlane/build.pl
Feel free to adapt it in any way necessary

agorararmard, 2021-01-14:
@thesourcerer8: Can you adopt the following:
1. Add a license header.
2. Modify the script to take a run directory (a directory), and print the logs with the numerical order they are presented in (in develop), i.e. 1-logX.log 2-logY.log 3-logX.log.
3. The script should take 2 arguments: 1st is the run directory, 2nd is the output file name.

donn, 2021-06-16:
Stale, closed, if you want to integrate something in the future consider a PR.

closed 2021-06-16 by donn